elastic/Machinebeat #4: MQTT module: SSL connection fails

packnology, 2019-11-04:
Sorry, not sure if this is the right place to post this. I have added a topic to the Elastic forum on this issue here: https://discuss.elastic.co/t/machinebeat-mqtt-module-ssl-connection-to-broker-fails/206369 . Please let me know if this might be a bug in the module.

felix-lessoer, 2019-11-05:
@packnology Thanks for raising the issue. I will check this. The SSL has only be tested with Cloud Service providers so far. Will create another test with mosquitto broker.
I assume it does work without SSL?

packnology, 2019-11-06:
Thanks, yes it does work without SSL. I was able to bridge the SSL to
non-SSL with an intermediary MQTT broker.

felix-lessoer, 2019-11-06:
@packnology I believe I found the issue you ran into. In the current version you have to add the protocol to the host parameter. Otherwise it assumes that you are using non SSL connection. Even if you set ssl: true.

e.g. host: "tcps://test.mosquitto.org:8883" works fine, but host: "test.mosquitto.org:8883" does not.
I will add a check for this behavior.

Let me know if this solves also your issue.

packnology, 2019-11-06:
Thanks Felix, I added the protocol to the host and it worked perfectly.
Thanks for your help!

closed 2019-12-07 by felix-lessoer

pratheekreddy, 2022-07-26:
once we configure TLS/SSL in mqtt config, can we test it with MQTT lens?