elastic/Machinebeat #5: OPC UA Connection

shettydprasad, 2019-11-15:
I am testing machine beat for OPC UA communication for kepware opc UA server data with elasticsearch. I am successfully transferred one tag data to elastic search. But, If I configure more than one tag only last tag configured in yml file is transferring to elastic. Please help me out to solve this issue. Please find attached yml file.
[opcua.zip](https://github.com/elastic/Machinebeat/files/3850734/opcua.zip)

PredictiveDataScience, 2019-12-06:
Hallo shettydprasad,
How did you solved this issue? Because I have the same problem with machinebeat. If I configure more than one tag, only the last one tag from config file yml is transferred to elasticsearch.
Thank you so much.
Lukas

shettydprasad, 2019-12-06:
Hello Lukas
The stated problem is still exists..! I thought i am configuring some thing wrong So i closed issue.! I used MQTT yml file for my testing instead OPC UA. 
I am reopening this issue..!

felix-lessoer, 2019-12-07:
@shettydprasad & @PredictiveDataScience : It should work now with multiple elements in the array. Can you verify, pls?

PredictiveDataScience, 2019-12-08:
Hello @felix-lessoer,
I downloaded and tested a new build version of the module (7.5) for windows,  but the issue still persists...
I downloaded machinebeat 7.5 from this link:
https://drive.google.com/file/d/15tpgO0Z1pT948mprovtOdTcSQNt5VCY2/view
I can´t see any changes in builded files (in github master branch changes were made)... but file date modified of machinebeat.exe in builded version is November...
Please can you rebuild (make) the new version of machinebeat 7.5?
Thank you so much.
Lukas.

felix-lessoer, 2019-12-09:
@PredictiveDataScience Thanks for letting me know. Took the wrong binary for the windows build. Now the windows version should be working as well.

PredictiveDataScience, 2019-12-09:
Hello @felix-lessoer ,
I tested the newest version, but there is a new error. I used the same configuration files (machinebeat.yml and opcua.yml) and tested 2 OPC UA endpoints, but the same error. (The old version with this endpoint worked well). Please check it again. 
Thank you so much.

```
019-12-09T10:08:34.656+0100    ERROR   nodevalue/nodevalue.go:119      Tried to connect to OPC UA server 1 time(s). Without success.
2019-12-09T10:08:34.687+0100    ERROR   instance/beat.go:877    Exiting: 1 error: [OPCUA] Failed to find suitable endpoint
Exiting: 1 error: [OPCUA] Failed to find suitable endpoint

```

opcua.yml configuration
```
- module: opcua
  metricsets: ["nodevalue"]
  enabled: true
  period: 2s

  endpoint: "opc.tcp://opcuaserver.com:48010"
  
  #subscribe: true
  #retryOnError: 5
  #maxThreads: 50

  nodes:
  -  ns: 3
     id: "AirConditioner_1.State"
     label: "State AirConditioner_1"
  -  ns: 3
     id: "AirConditioner_1.Humidity"
     label: "Humidity AirConditioner_1"
  -  ns: 3
     id: "AirConditioner_1.Temperature"
     label: "Temperature AirConditioner_1"
```

felix-lessoer, 2019-12-09:
This is including security capabilities now. 
Try to set policy: "" like this to ignore the security policies.

```
- module: opcua
  metricsets: ["nodevalue"]
  enabled: true
  period: 2s

  endpoint: "opc.tcp://opcuaserver.com:48010"
  policy: ""

  #subscribe: true
  #retryOnError: 5
  #maxThreads: 50

  nodes:
  -  ns: 3
     id: "AirConditioner_1.State"
     label: "State AirConditioner_1"
  -  ns: 3
     id: "AirConditioner_1.Humidity"
     label: "Humidity AirConditioner_1"
  -  ns: 3
     id: "AirConditioner_1.Temperature"
     label: "Temperature AirConditioner_1"
```

PredictiveDataScience, 2019-12-09:
Thank you @felix-lessoer,
everything works fine.  
It works now with multiple elements in the array and also connection to endpoint works great with `policy: ""`
Issues is solved.
I look forward to more features.
Best regards, 
Lukas

shettydprasad, 2019-12-10:
Its working as expected.. Thank you

closed 2019-12-10 by shettydprasad