elastic/Machinebeat #7: cannot find suitable endpoint error

markusTh83, 2019-12-13:
Hi,

I am trying to connect to an OPC-UA Server, but without success. Both when I try connecting via IP or (string) Hostname, I get an '_cannot find suitable endpoint_' error:

`PS C:\machinebeat\machine> .\machinebeat.exe -e -c machinebeat.yml -d "*"

2019-12-13T10:16:43.972+0100    INFO    instance/beat.go:606    Home path: [C:\machinebeat\machine] Config path: [C:\machinebeat\machine] Data path: [C:\machinebeat\machine\data] Logs path: [C:\machinebeat\machine\logs]
2019-12-13T10:16:43.973+0100    DEBUG   [beat]  instance/beat.go:658    Beat metadata path: C:\machinebeat\machine\data\meta.json
2019-12-13T10:16:43.974+0100    INFO    instance/beat.go:614    Beat ID: 4db98739-3d2b-4be2-bc75-0449e2469e59
2019-12-13T10:16:43.983+0100    DEBUG   [filters]       add_cloud_metadata/add_cloud_metadata.go:164    add_cloud_metadata: starting to fetch metadata, timeout=3s
2019-12-13T10:16:43.986+0100    DEBUG   [filters]       add_cloud_metadata/add_cloud_metadata.go:196    add_cloud_metadata: received disposition for qcloud after 3.0003ms. result=[provider:qcloud, error=failed requesting qcloud metadata: Ge
t http://metadata.tencentyun.com/meta-data/placement/zone: dial tcp: lookup metadata.tencentyun.com: no such host, metadata={}]
2019-12-13T10:16:44.055+0100    DEBUG   [filters]       add_cloud_metadata/add_cloud_metadata.go:196    add_cloud_metadata: received disposition for ecs after 71.3372ms. result=[provider:ecs, error=failed with http status code 401, metadata={}]
2019-12-13T10:16:46.995+0100    DEBUG   [filters]       add_cloud_metadata/add_cloud_metadata.go:196    add_cloud_metadata: received disposition for aws after 3.0111452s. result=[provider:aws, error=failed requesting aws metadata: Get http://169.254.169.254/2014-02-25/dynamic/instance-identity/document: net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers), metadata={}]
2019-12-13T10:16:46.995+0100    DEBUG   [filters]       add_cloud_metadata/add_cloud_metadata.go:196    add_cloud_metadata: received disposition for digitalocean after 3.0111452s. result=[provider:digitalocean, error=failed requesting digitalocean metadata: Get http://169.254.169.254/metadata/v1.json: net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers), metadata={}]
2019-12-13T10:16:46.996+0100    DEBUG   [filters]       add_cloud_metadata/add_cloud_metadata.go:196    add_cloud_metadata: received disposition for gcp after 3.0127665s. result=[provider:gcp, error=failed requesting gcp metadata: Get http://169.254.169.254/computeMetadata/v1/?recursive=true&alt=json: net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers), metadata={}]
2019-12-13T10:16:46.997+0100    DEBUG   [filters]       add_cloud_metadata/add_cloud_metadata.go:203    add_cloud_metadata: timed-out waiting for all responses
2019-12-13T10:16:46.997+0100    DEBUG   [filters]       add_cloud_metadata/add_cloud_metadata.go:167    add_cloud_metadata: fetchMetadata ran for 3.0137625s
2019-12-13T10:16:46.997+0100    INFO    add_cloud_metadata/add_cloud_metadata.go:347    add_cloud_metadata: hosting provider type not detected.
2019-12-13T10:16:46.997+0100    DEBUG   [processors]    processors/processor.go:101     Generated new processors: add_host_metadata=[netinfo.enabled=[false], cache.ttl=[5m0s]], add_cloud_metadata=null
2019-12-13T10:16:46.998+0100    DEBUG   [seccomp]       seccomp/seccomp.go:96   Syscall filtering is only supported on Linux
2019-12-13T10:16:46.998+0100    INFO    [beat]  instance/beat.go:902    Beat info       {"system_info": {"beat": {"path": {"config": "C:\\machinebeat\\machine", "data": "C:\\machinebeat\\machine\\data", "home": "C:\\machinebeat\\machine", "logs": "C:\\machinebeat\\machine\\logs"}, "type": "machinebeat", "uuid": "4db98739-3d2b-4be2-bc75-0449e2469e59"}}}
2019-12-13T10:16:46.999+0100    INFO    [beat]  instance/beat.go:911    Build info      {"system_info": {"build": {"commit": "unknown", "libbeat": "8.0.0", "time": "1754-08-30T22:43:41.128Z", "version": "8.0.0"}}}
2019-12-13T10:16:46.999+0100    INFO    [beat]  instance/beat.go:914    Go runtime info {"system_info": {"go": {"os":"windows","arch":"amd64","max_procs":2,"version":"go1.12.4"}}}
2019-12-13T10:16:47.004+0100    INFO    [beat]  instance/beat.go:918    Host info       {"system_info": {"host": {"architecture":"x86_64","boot_time":"2019-04-01T08:47:21.39+02:00","name":"myBeat","ip":["fe80::b878:f783:db9:a3df/64","2.3.4.5/16","::1/128","127.0.0.1/8","fe80::5efe:ac1e:173/128"],"kernel_version":"6.3.9600.19153 (winblue_ltsb.180908-0600)","mac":["00:50:56:91:07:1c","00:00:00:00:00:00:00:e0"],"os":{"family":"windows","platform":"windows","name":"Windows Server 2012 R2 Datacenter","version":"6.3","major":3,"minor":0,"patch":0,"build":"9600.19155"},"timezone":"CET","timezone_offset_sec":3600,"id":"3a43f2c3-68d2-47b0-bd20-bbc5831cf6c0"}}}
2019-12-13T10:16:47.006+0100    INFO    [beat]  instance/beat.go:947    Process info    {"system_info": {"process": {"cwd": "C:\\machinebeat\\machine", "exe": "C:\\machinebeat\\machine\\machinebeat.exe", "name": "machinebeat.exe", "pid": 34340, "ppid": 22244, "start_time": "2019-12-13T10:16:43.094+0100"}}}
2019-12-13T10:16:47.006+0100    INFO    instance/beat.go:292    Setup Beat: machinebeat; Version: 8.0.0
2019-12-13T10:16:47.006+0100    DEBUG   [beat]  instance/beat.go:318    Initializing output plugins
2019-12-13T10:16:47.006+0100    INFO    [index-management]      idxmgmt/std.go:178      Set output.elasticsearch.index to 'machinebeat-8.0.0' as ILM is enabled.
2019-12-13T10:16:47.007+0100    INFO    elasticsearch/client.go:170     Elasticsearch url: http://elastic:9200
2019-12-13T10:16:47.008+0100    DEBUG   [publisher]     pipeline/consumer.go:137        start pipeline event consumer
2019-12-13T10:16:47.008+0100    INFO    [publisher]     pipeline/module.go:97   Beat name: myBeat
2019-12-13T10:16:47.009+0100    DEBUG   [modules]       beater/metricbeat.go:121        Available modules and metricsets: Register [ModuleFactory:[beat, docker, elasticsearch, kibana, logstash, mongodb, mysql, postgresql, system, uwsgi, windows], MetricSetFactory:[aerospike/namespace, apache/status, beat/state, beat/stats, ceph/cluster_disk, ceph/cluster_health, ceph/cluster_status, ceph/monitor_health, ceph/osd_df, ceph/osd_tree, ceph/pool_disk, consul/agent, couchbase/bucket, couchbase/cluster, couchbase/node, couchdb/server, docker/container, docker/cpu, docker/diskio, docker/event, docker/healthcheck, docker/image, docker/info, docker/memory, docker/network, dropwizard/collector, elasticsearch/ccr, elasticsearch/cluster_stats, elasticsearch/index, elasticsearch/index_recovery, elasticsearch/index_summary, elasticsearch/ml_job, elasticsearch/node, elasticsearch/node_stats, elasticsearch/pending_tasks, elasticsearch/shard, envoyproxy/server, etcd/leader, etcd/metrics, etcd/self, etcd/store, golang/expvar, golang/heap, graphite/server, haproxy/info, haproxy/stat, http/json, http/server, jolokia/jmx, kafka/consumergroup, kafka/partition, kibana/stats, kibana/status, kubernetes/apiserver, kubernetes/container, kubernetes/controllermanager, kubernetes/event, kubernetes/node, kubernetes/pod, kubernetes/proxy, kubernetes/scheduler, kubernetes/state_container, kubernetes/state_deployment, kubernetes/state_node, kubernetes/state_pod, kubernetes/state_replicaset, kubernetes/state_statefulset, kubernetes/system, kubernetes/volume, kvm/dommemstat, logstash/node, logstash/node_stats, memcached/stats, mongodb/collstats, mongodb/dbstats, mongodb/metrics, mongodb/replstatus, mongodb/status, mqtt/topic, munin/node, mysql/galera_status, mysql/status, nats/connections, nats/routes, nats/stats, nats/subscriptions, nginx/stubstatus, opcua/nodevalue, php_fpm/pool, php_fpm/process, postgresql/activity, postgresql/bgwriter, postgresql/database, postgresql/statement, prometheus/collector, rabbitmq/connection, rabbitmq/exchange, rabbitmq/node, rabbitmq/queue, redis/info, redis/key, redis/keyspace, system/core, system/cpu, system/diskio, system/filesystem, system/fsstat, system/memory, system/network, system/process, system/process_summary, system/raid, system/socket_summary, system/uptime, traefik/health, uwsgi/status, vsphere/datastore, vsphere/host, vsphere/virtualmachine, windows/perfmon, windows/service, zookeeper/connection, zookeeper/mntr, zookeeper/server]]
[... cut ...]
2019-12-13T10:17:09.073+0100    ERROR   instance/beat.go:877    Exiting: 1 error: [OPCUA] Failed to find suitable endpoint
Exiting: 1 error: [OPCUA] Failed to find suitable endpoint`

The yml file contains

`- module: opcua
  metricsets: ["nodevalue"]
  enabled: true
  period: 2s

  #The URL of your OPC UA Server
  endpoint: "opc.tcp://machine-PC:12345"
  #endpoint: "opc.tcp://1.2.3.4:12345"
  policy: ""
  `
and the node settings with IDs and labels.

Via UaExpert the connection is successful. Is this an issue or a config problem?

Kind regards,

Markus

felix-lessoer, 2019-12-13:
Hey Markus,

thanks for raising this issue.
Can you tell me which kind (implementation) of OPCUA server you are using?

Would like to reproduce that problem on my local machine.

felix-lessoer, 2019-12-13:
@markusTh83 Can you check the latest version? I added security now and improved the endpoint detection.

markusTh83, 2019-12-16:
@felix-lessoer 
I've updated my post from Friday, had an C&P error.
I will check the latest version, thougt I've already had it.

markusTh83, 2019-12-16:
I've downloaded the actual version and tried again (with the Server Name as string, the IP to string resolution fails with the same error):

```

PS C:\machinebeat\MyMachine> .\machinebeat.exe -e -c machinebeat.yml -d "*"
2019-12-16T09:45:55.616+0100    INFO    instance/beat.go:606    Home path: [C:\machinebeat\MyMachine] Config path: [C:\machinebeat\MyMachine] Data path: [C:\machinebeat\MyMachine\data] Logs path: [C:\machinebeat\MyMachine\logs]
2019-12-16T09:45:55.617+0100    DEBUG   [beat]  instance/beat.go:658    Beat metadata path: C:\machinebeat\MyMachine\data\meta.json
2019-12-16T09:45:55.618+0100    INFO    instance/beat.go:614    Beat ID: 9cf56f24-51d4-46c0-b251-196dbef38c7b
2019-12-16T09:45:55.626+0100    DEBUG   [filters]       add_cloud_metadata/add_cloud_metadata.go:164    add_cloud_metadata: starting to fetch metadata, timeout=3s
2019-12-16T09:45:55.630+0100    DEBUG   [filters]       add_cloud_metadata/add_cloud_metadata.go:196    add_cloud_metadata: received disposition for qcloud after 3.9987ms. result=[provider:qcloud, error=failed requesting qcloud metadata: Get http://metadata.tencentyun.com/meta-data/placement/region: dial tcp: lookup metadata.tencentyun.com: no such host, metadata={}]
2019-12-16T09:45:55.687+0100    DEBUG   [filters]       add_cloud_metadata/add_cloud_metadata.go:196    add_cloud_metadata: received disposition for ecs after 61.3743ms. result=[provider:ecs, error=failed with http status code 401, metadata={}]
2019-12-16T09:45:58.628+0100    DEBUG   [filters]       add_cloud_metadata/add_cloud_metadata.go:196    add_cloud_metadata: received disposition for openstack after 3.001865s. result=[provider:openstack, error=failed requesting openstack metadata: Get http://169.254.169.254/2009-04-04/meta-data/placement/availability-zone: net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers), metadata={}]
2019-12-16T09:45:58.628+0100    DEBUG   [filters]       add_cloud_metadata/add_cloud_metadata.go:203    add_cloud_metadata: timed-out waiting for all responses
2019-12-16T09:45:58.628+0100    DEBUG   [filters]       add_cloud_metadata/add_cloud_metadata.go:167    add_cloud_metadata: fetchMetadata ran for 3.001865s
2019-12-16T09:45:58.629+0100    INFO    add_cloud_metadata/add_cloud_metadata.go:347    add_cloud_metadata: hosting provider type not detected.
2019-12-16T09:45:58.629+0100    DEBUG   [processors]    processors/processor.go:101     Generated new processors: add_host_metadata=[netinfo.enabled=[false], cache.ttl=[5m0s]], add_cloud_metadata=null
2019-12-16T09:45:58.629+0100    DEBUG   [seccomp]       seccomp/seccomp.go:96   Syscall filtering is only supported on Linux
2019-12-16T09:45:58.629+0100    INFO    [beat]  instance/beat.go:902    Beat info       {"system_info": {"beat": {"path": {"config": "C:\\machinebeat\\MyMachine", "data": "C:\\machinebeat\\MyMachine\\data", "home": "C:\\machinebeat\\MyMachine", "logs": "C:\\machinebeat\\MyMachine\\logs"}, "type": "machinebeat", "uuid": "9cf56f24-51d4-46c0-b251-196dbef38c7b"}}}
2019-12-16T09:45:58.630+0100    INFO    [beat]  instance/beat.go:911    Build info      {"system_info": {"build": {"commit": "unknown", "libbeat": "8.0.0", "time": "1754-08-30T22:43:41.128Z", "version": "8.0.0"}}}
2019-12-16T09:45:58.630+0100    INFO    [beat]  instance/beat.go:914    Go runtime info {"system_info": {"go": {"os":"windows","arch":"amd64","max_procs":2,"version":"go1.12.4"}}}
2019-12-16T09:45:58.635+0100    INFO    [beat]  instance/beat.go:918    Host info       {"system_info": {"host": {"architecture":"x86_64","boot_time":"2019-04-01T08:47:25.11+02:00","name":"myBeat","ip":["fe80::b878:f783:db9:a3df/64","172.30.1.115/16","::1/128","127.0.0.1/8","fe80::5efe:ac1e:173/128"],"kernel_version":"6.3.9600.19153 (winblue_ltsb.180908-0600)","mac":["00:50:56:91:07:1c","00:00:00:00:00:00:00:e0"],"os":{"family":"windows","platform":"windows","name":"Windows Server 2012 R2 Datacenter","version":"6.3","major":3,"minor":0,"patch":0,"build":"9600.19155"},"timezone":"CET","timezone_offset_sec":3600,"id":"3a43f2c3-68d2-47b0-bd20-bbc5831cf6c0"}}}
2019-12-16T09:45:58.637+0100    INFO    [beat]  instance/beat.go:947    Process info    {"system_info": {"process": {"cwd": "C:\\machinebeat\\MyMachine", "exe": "C:\\machinebeat\\machinebeat\\machinebeat.exe", "name": "machinebeat.exe", "pid": 25928, "ppid": 22244, "start_time": "2019-12-16T09:45:54.711+0100"}}}
2019-12-16T09:45:58.638+0100    INFO    instance/beat.go:292    Setup Beat: machinebeat; Version: 8.0.0
2019-12-16T09:45:58.638+0100    DEBUG   [beat]  instance/beat.go:318    Initializing output plugins
2019-12-16T09:45:58.638+0100    INFO    [index-management]      idxmgmt/std.go:178      Set output.elasticsearch.index to 'machinebeat-8.0.0' as ILM is enabled.
2019-12-16T09:45:58.638+0100    INFO    elasticsearch/client.go:170     Elasticsearch url: http://myElasticStack:9200
2019-12-16T09:45:58.639+0100    DEBUG   [publisher]     pipeline/consumer.go:137        start pipeline event consumer
2019-12-16T09:45:58.639+0100    INFO    [publisher]     pipeline/module.go:97   Beat name: myBeat
2019-12-16T09:45:58.639+0100    DEBUG   [modules]       beater/metricbeat.go:121        Available modules and metricsets: Register [ModuleFactory:[beat, docker, elasticsearch, kibana, logstash, mongodb, mysql, postgresql, system, uwsgi, windows], MetricSetFactory:[aerospike/namespace, apache/status, beat/state, beat/stats, ceph/cluster_disk, ceph/cluster_health,  [... cut ...] c/beats/libbeat/cmd/instance/beat.go:431\ngithub.com/elastic/beats/libbeat/cmd/instance.Run.func1\n\tC:/Users/felix/projekte/machinebeat/src/github.com/elastic/beats/libbeat/cmd/instance/beat.go:182\ngithub.com/elastic/beats/libbeat/cmd/instance.Run\n\tC:/Users/felix/projekte/machinebeat/src/github.com/elastic/beats/libbeat/cmd/instance/beat.go:183\nmain.main\n\tC:/Users/felix/projekte/machinebeat/src/github.com/elastic/machinebeat/main.go:21\nruntime.main\n\tc:/go/src/runtime/proc.go:200"}

```

felix-lessoer, 2020-03-18:
This happens when the certificates are not set according to the setup of the server.
Be careful to set policy, mode and certification info based on your server settings.

This is an example configuration incl. security:
```
# Module: opcua
# Docs: https://www.elastic.co/guide/en/beats/machinebeat//machinebeat-module-opcua.html

- module: opcua
  metricsets: ["nodevalue"]
  enabled: true
  period: 2s
  
  #The URL of your OPC UA Server
  endpoint: "opc.tcp://milo.digitalpetri.com:62541/milo"
  username: "user1"
  password: "password"
  #policy: "none"
  #mode: "none"
  #clientCert: ""
  #clientKey: ""
  
  ##The subscribe mode collects one event whenever there is a change of the value.
  ## All collected events will be send after the configured period
  ## If you turn this off the beat will pull the current value after each period.
  #subscribe: true
  
  ##How often we should retry to connect to the opcua server
  #retryOnError: 5

  ##How many collect threads should be opened. This is only relevant for non subscription mode.  
  ## This limit can be reached when it takes longer to get a value than it is configured with period. 
  #maxThreads: 50

  nodes:
  -  ns: 2
     id: "Dynamic/RandomDouble"
     label: "Random Double"

```

closed 2020-03-18 by felix-lessoer