elastic/Machinebeat #8: opcua module label not set correctly

cskowronnek, 2020-04-07:
```
  nodes: 
  -  id: "ns=1;i=176000"
     label: "Zentrale - Summe der Wirkleistung"
  -  id: "ns=1;i=176080"
     label: "Zentrale - Temp_Umgebung_1"
  -  id: "ns1;i=176048"
     label: "Zentrale - Wirkarbeit"
```

Thats my node configuration in opcua module because the discovery does not work. However, only the last label is applied to all of the nodes. Is this a bug?

Kind regards
Conrad

felix-lessoer, 2020-04-07:
Hi Conrad,

I will check this label thing locally.
Can you tell me in the meantime why the node discovery is not working for you?

Would be nice to have some logs. The discovery mode is a new feature .. 

Thanks
Felix

cskowronnek, 2020-04-07:
Hi Felix,

I think discovery does not work because it can't find endpoints. The Loytec Gateways has a local WAN IP and I access it via Public IP + NAT. 
When I connect using UaExpert a message appears, that UaExpert replaces the private ip with the public ip to find the endpoints. I'll try it in the local network tomorrow and will report back if it works there.

cskowronnek, 2020-04-07:
regarding discovery:

```
2020-04-07T16:12:40.413Z        INFO    nodevalue/client.go:95  [OPCUA] Connection established
2020-04-07T16:12:40.413Z        INFO    nodevalue/client.go:282 [OPCUA] Start browsing from node: root
2020-04-07T16:12:40.508Z        INFO    nodevalue/client.go:312 Analyse node id i=85
2020-04-07T16:12:40.508Z        DEBUG   [Get DataType]  nodevalue/client.go:373 Status not okay
```

felix-lessoer, 2020-04-07:
I've uploaded a new version. For me the labels are now working fine.

The discovery is also working fine. The last log line you shared is expected. There should be something more after that. Could you share more?

cskowronnek, 2020-04-08:
```
2020-04-08T11:45:02.256+0200    INFO    instance/beat.go:439    machinebeat start running.
2020-04-08T11:45:02.261+0200    DEBUG   [cfgfile]       cfgfile/reload.go:133   Checking module configs from: C:Downloads\machinebeat/modules.d/*.yml
2020-04-08T11:45:02.256+0200    INFO    [monitoring]    log/log.go:118  Starting metrics logging every 30s
2020-04-08T11:45:02.256+0200    DEBUG   [service]       service/service_windows.go:72   Windows is interactive: true
2020-04-08T11:45:02.263+0200    DEBUG   [cfgfile]       cfgfile/cfgfile.go:193  Load config from file: C:\Downloads\machinebeat\modules.d\opcua.yml
2020-04-08T11:45:02.266+0200    DEBUG   [cfgfile]       cfgfile/reload.go:147   Number of module configs found: 1
2020-04-08T11:45:02.267+0200    WARN    [cfgwarn]       nodevalue/nodevalue.go:89       BETA: The OPCUA metricset is beta.
2020-04-08T11:45:02.268+0200    INFO    nodevalue/client.go:51  [OPCUA] Get all endpoints from opc.tcp://192.168.0.110:4172
2020-04-08T11:45:02.282+0200    INFO    nodevalue/client.go:64  [OPCUA] Policy URI: http://opcfoundation.org/UA/SecurityPolicy#None with security mode MessageSecurityModeNone
2020-04-08T11:45:02.283+0200    INFO    nodevalue/client.go:80  [OPCUA] Set authentication information
2020-04-08T11:45:02.284+0200    INFO    nodevalue/client.go:81  [OPCUA] User: operator
2020-04-08T11:45:02.437+0200    INFO    nodevalue/client.go:95  [OPCUA] Connection established
2020-04-08T11:45:02.437+0200    INFO    nodevalue/client.go:283 [OPCUA] Start browsing from node: root
2020-04-08T11:45:02.450+0200    INFO    nodevalue/client.go:313 Analyse node id i=85
2020-04-08T11:45:02.450+0200    DEBUG   [Get DataType]  nodevalue/client.go:374 Status not okay
2020-04-08T11:45:02.454+0200    INFO    cfgfile/reload.go:171   Config reloader started
2020-04-08T11:45:02.461+0200    DEBUG   [cfgfile]       cfgfile/reload.go:197   Scan for new config files
2020-04-08T11:45:02.464+0200    DEBUG   [cfgfile]       cfgfile/cfgfile.go:193  Load config from file: C:\Downloads\machinebeat\modules.d\opcua.yml
2020-04-08T11:45:02.474+0200    DEBUG   [cfgfile]       cfgfile/reload.go:216   Number of module configs found: 1
2020-04-08T11:45:02.477+0200    DEBUG   [reload]        cfgfile/list.go:62      Starting reload procedure, current runners: 0
2020-04-08T11:45:02.480+0200    DEBUG   [reload]        cfgfile/list.go:80      Start list: 1, Stop list: 0
2020-04-08T11:45:02.483+0200    WARN    [cfgwarn]       nodevalue/nodevalue.go:89       BETA: The OPCUA metricset is beta.
2020-04-08T11:45:02.484+0200    WARN    nodevalue/nodevalue.go:119      A new connection attempt was made. This gets ignored from this module
2020-04-08T11:45:02.494+0200    DEBUG   [reload]        cfgfile/list.go:101     Starting runner: opcua [metricsets=1]
2020-04-08T11:45:02.496+0200    DEBUG   [module]        module/wrapper.go:120   Starting Wrapper[name=opcua, len(metricSetWrappers)=1]
2020-04-08T11:45:02.497+0200    INFO    cfgfile/reload.go:226   Loading of config files completed.
2020-04-08T11:45:02.497+0200    DEBUG   [module]        module/wrapper.go:182   Starting metricSetWrapper[module=opcua, name=nodevalue, host=]
2020-04-08T11:45:02.507+0200    INFO    nodevalue/nodevalue.go:168      [OPCUA] Publishing 0 new events
```

Just tried it locally and it didn't find anything. I tried with the zip-release and not with your latest commits. I'll set up a build environment later and try your patch

felix-lessoer, 2020-04-08:
In the zip file there is already the latest version. So no need to set up your own environment.
The problem with the discovery is that it tries to find sth. in the Object folder but this seems to be empty. The object folder is the one the beat takes as root.
You can also use any root folder of your choice. The browsing will start from that point.

cskowronnek, 2020-04-08:
I tried several different subfolders and there is an Object folder with node id 85. I could not see any anomalies when using UAexpert

felix-lessoer, 2020-04-08:
Well, thats definitely not the expected behavior. 
But can you confirm that the subscription to the nodes works . Also with the right labeling of the nodes in Kibana?

cskowronnek, 2020-04-08:
Subscription to explicitly configured nodes works fine and the labels are now correct too.

closed 2020-04-08 by cskowronnek