elastic/Machinebeat #9: opcua.nodevalue.value can only handle on data type

cskowronnek, 2020-04-07:
I read several numeric values from our OPC-UA gateway which works fine. All values get written on opcua.nodevalue.value and I can filter bei node id or  opcua.node.Name.
If add a node which has a non-numerical data type (e.g. string) it won't index the data because the Elastic mapping for opcua.nodevalue.value is already set to type:long. 

It would be good to have a way to handle different data types

felix-lessoer, 2020-04-07:
I now changed the name of the value field. The field name contains the data type. This way the automatic elastic mapping will work fine even with different data types.

Is it now working for you?

cskowronnek, 2020-04-08:
Data types are now on individual fields. I have int32 and float64 at the moment and will test with string later.

cskowronnek, 2020-04-09:
String works also fine

closed 2020-04-09 by cskowronnek