elastic/Machinebeat #11: opcua: subscribe=false leads to ES error

cskowronnek, 2020-04-09:
We need the data time continuously through time so I set subscribe to false. When Logstash tries to write the data to ES I get an error.

`[2020-04-08T11:18:56,291][WARN ][logstash.outputs.elasticsearch] Could not index event to Elasticsearch. {:status=>400, :action=>["index", {:_id=>nil, :_index=>"machinebeat-opcua-2020.04",
:routing=>nil, :_type=>"_doc"}, #<LogStash::Event:0x450bf4f9>], :response=>{"index"=>{"_index"=>"machinebeat-opcua-2020.04", "_type"=>"_doc", "_id"=>"aT6DWXEBX62O6uTftSed", "status"=>400, "
error"=>{"type"=>"mapper_parsing_exception", "reason"=>"failed to parse", "caused_by"=>{"type"=>"illegal_argument_exception", "reason"=>"field name cannot be an empty string"}}}}}`

I wrote the logstash events to a text file but can't spot any errors

felix-lessoer, 2020-04-09:
Oh this is probably a issue based on the change with the data type in field name. The data type is not extracted in the non subscribe mode. As this is a different method in OPC UA Framework.

cskowronnek, 2020-04-09:
```
         "opcua" => {
        "nodevalue" => {
                   "" => "192.168.1.254",
            "created" => "2020-04-08 09:54:49.384758 +0000 UTC",
              "state" => "OK"
        },
```
Yep I found it now. The field name is empty

felix-lessoer, 2020-04-12:
This is fixed now.

cskowronnek, 2020-04-14:
It works now. Thank you very much

closed 2020-04-14 by cskowronnek