elastic/Machinebeat #12: OPC-UA browsing doesn't work

cskowronnek, 2020-04-14:
I reopen this as a new issue after discussing it in another issue. Browsing does not work on our Gateway. Subscribing to explicitly configured nodes works and browsing also works in UAexpert.

Debug log:
```
2020-04-14T08:02:09.469Z        INFO    instance/beat.go:439    machinebeat start running.
2020-04-14T08:02:09.469Z        DEBUG   [cfgfile]       cfgfile/reload.go:133   Checking module configs from: /home/cleaned/machinebeat-test/modules.d/*.yml
2020-04-14T08:02:09.469Z        DEBUG   [cfgfile]       cfgfile/cfgfile.go:193  Load config from file: /home/cleaned/machinebeat-test/modules.d/opcua.yml
2020-04-14T08:02:09.469Z        DEBUG   [cfgfile]       cfgfile/reload.go:147   Number of module configs found: 1
2020-04-14T08:02:09.469Z        WARN    [cfgwarn]       nodevalue/nodevalue.go:92       BETA: The OPCUA metricset is beta.
2020-04-14T08:02:09.469Z        INFO    nodevalue/client.go:51  [OPCUA] Get all endpoints from opc.tcp://cleaned:cleaned
2020-04-14T08:02:13.162Z        INFO    nodevalue/client.go:64  [OPCUA] Policy URI: http://opcfoundation.org/UA/SecurityPolicy#None with security mode MessageSecurityModeNone
2020-04-14T08:02:13.162Z        INFO    nodevalue/client.go:80  [OPCUA] Set authentication information
2020-04-14T08:02:13.162Z        INFO    nodevalue/client.go:81  [OPCUA] User: cleaned
2020-04-14T08:02:20.725Z        INFO    nodevalue/client.go:95  [OPCUA] Connection established
2020-04-14T08:02:20.725Z        INFO    nodevalue/client.go:297 [OPCUA] Start browsing from node: root
2020-04-14T08:02:21.785Z        INFO    nodevalue/client.go:327 Analyse node id i=85
2020-04-14T08:02:21.785Z        DEBUG   [Get DataType]  nodevalue/client.go:388 Status not okay
2020-04-14T08:02:22.125Z        INFO    cfgfile/reload.go:171   Config reloader started
2020-04-14T08:02:22.125Z        DEBUG   [cfgfile]       cfgfile/reload.go:197   Scan for new config files
2020-04-14T08:02:22.125Z        DEBUG   [cfgfile]       cfgfile/cfgfile.go:193  Load config from file: /home/cleaned/machinebeat-test/modules.d/opcua.yml
2020-04-14T08:02:22.126Z        DEBUG   [cfgfile]       cfgfile/reload.go:216   Number of module configs found: 1
2020-04-14T08:02:22.126Z        DEBUG   [reload]        cfgfile/list.go:62      Starting reload procedure, current runners: 0
2020-04-14T08:02:22.126Z        DEBUG   [reload]        cfgfile/list.go:80      Start list: 1, Stop list: 0
2020-04-14T08:02:22.126Z        WARN    [cfgwarn]       nodevalue/nodevalue.go:92       BETA: The OPCUA metricset is beta.
2020-04-14T08:02:22.126Z        WARN    nodevalue/nodevalue.go:124      A new connection attempt was made. This gets ignored from this module
2020-04-14T08:02:22.126Z        DEBUG   [reload]        cfgfile/list.go:101     Starting runner: opcua [metricsets=1]
2020-04-14T08:02:22.126Z        DEBUG   [module]        module/wrapper.go:120   Starting Wrapper[name=opcua, len(metricSetWrappers)=1]
2020-04-14T08:02:22.126Z        INFO    cfgfile/reload.go:226   Loading of config files completed.
2020-04-14T08:02:22.126Z        DEBUG   [module]        module/wrapper.go:182   Starting metricSetWrapper[module=opcua, name=nodevalue, host=]
2020-04-14T08:02:22.127Z        INFO    nodevalue/nodevalue.go:185      [OPCUA] Publishing 0 new events
2020-04-14T08:02:25.127Z        INFO    nodevalue/nodevalue.go:185      [OPCUA] Publishing 0 new events
2020-04-14T08:02:28.127Z        INFO    nodevalue/nodevalue.go:185      [OPCUA] Publishing 0 new events
2020-04-14T08:02:31.127Z        INFO    nodevalue/nodevalue.go:185      [OPCUA] Publishing 0 new events
2020-04-14T08:02:34.127Z        INFO    nodevalue/nodevalue.go:185      [OPCUA] Publishing 0 new events
2020-04-14T08:02:34.127Z        INFO    nodevalue/nodevalue.go:176      [OPCUA] Too much zero publish attempts.
```

felix-lessoer, 2020-04-14:
Could you share more information about the OPC UA Server implementation that you are using? Using my test server everything works as expected.

cskowronnek, 2020-04-14:
It's a Loytec LGATE-902

felix-lessoer, 2020-04-15:
I changed the browsing parameter a bit in the latest version. Could you check how it works now for your OPC UA Server?

cskowronnek, 2020-04-16:
I still see the same behavior with 7.6.3

felix-lessoer, 2020-04-16:
I also added additional debug logging. Could you share debug logs?

cskowronnek, 2020-04-17:
```
2020-04-17T07:11:45.693Z        DEBUG   [cfgfile]       cfgfile/reload.go:147   Number of module configs found: 1
2020-04-17T07:11:45.693Z        WARN    [cfgwarn]       nodevalue/nodevalue.go:92       BETA: The OPCUA metricset is beta.
2020-04-17T07:11:45.693Z        INFO    nodevalue/client.go:51  [OPCUA] Get all endpoints from opc.tcp://123.123.123.123:123
2020-04-17T07:11:45.693Z        INFO    [monitoring]    log/log.go:118  Starting metrics logging every 30s
2020-04-17T07:11:46.275Z        INFO    nodevalue/client.go:64  [OPCUA] Policy URI: http://opcfoundation.org/UA/SecurityPolicy#None with security mode MessageSecurityModeNone
2020-04-17T07:11:46.275Z        INFO    nodevalue/client.go:80  [OPCUA] Set authentication information
2020-04-17T07:11:46.275Z        INFO    nodevalue/client.go:81  [OPCUA] User: user
2020-04-17T07:11:47.157Z        INFO    nodevalue/client.go:95  [OPCUA] Connection established
2020-04-17T07:11:47.158Z        INFO    nodevalue/client.go:299 [OPCUA] Start browsing from node: root
2020-04-17T07:11:47.158Z        DEBUG   [Browse]        nodevalue/client.go:318 Start browsing
2020-04-17T07:11:47.281Z        INFO    nodevalue/client.go:330 Analyse node id i=85
2020-04-17T07:11:47.282Z        DEBUG   [Get DataType]  nodevalue/client.go:397 Status not okay
2020-04-17T07:11:47.396Z        DEBUG   [Browse]        nodevalue/client.go:360 The view timestamp is not available or not supported. StatusBadViewTimestampInvalid (0x80C90000)
2020-04-17T07:11:47.397Z        INFO    cfgfile/reload.go:171   Config reloader started
2020-04-17T07:11:47.397Z        DEBUG   [cfgfile]       cfgfile/reload.go:197   Scan for new config files
2020-04-17T07:11:47.397Z        DEBUG   [cfgfile]       cfgfile/cfgfile.go:193  Load config from file: /home/cleaned/machinebeat-test/modules.d/opcua.yml
2020-04-17T07:11:47.397Z        DEBUG   [cfgfile]       cfgfile/reload.go:216   Number of module configs found: 1
2020-04-17T07:11:47.397Z        DEBUG   [reload]        cfgfile/list.go:62      Starting reload procedure, current runners: 0
2020-04-17T07:11:47.398Z        DEBUG   [reload]        cfgfile/list.go:80      Start list: 1, Stop list: 0
2020-04-17T07:11:47.398Z        WARN    [cfgwarn]       nodevalue/nodevalue.go:92       BETA: The OPCUA metricset is beta.
2020-04-17T07:11:47.398Z        WARN    nodevalue/nodevalue.go:124      A new connection attempt was made. This gets ignored from this module
2020-04-17T07:11:47.398Z        DEBUG   [reload]        cfgfile/list.go:101     Starting runner: opcua [metricsets=1]
2020-04-17T07:11:47.398Z        DEBUG   [module]        module/wrapper.go:120   Starting Wrapper[name=opcua, len(metricSetWrappers)=1]
2020-04-17T07:11:47.398Z        INFO    cfgfile/reload.go:226   Loading of config files completed.
2020-04-17T07:11:47.398Z        DEBUG   [module]        module/wrapper.go:182   Starting metricSetWrapper[module=opcua, name=nodevalue, host=]
2020-04-17T07:11:47.398Z        INFO    nodevalue/nodevalue.go:185      [OPCUA] Publishing 0 new events
```

felix-lessoer, 2020-04-17:
This is a helpful log. Thanks. I will try to find a solution.

cskowronnek, 2020-04-22:
I also tried with endpoint: "opc.tcp://opcuademo.sterfive.com:26543", which works fine.
But on server  "opc.tcp://opcua.rocks:4840/" I get this. It looks like the endpoint points to localhost:4840. Many OPCUA servers show this gap between public IP and internal hostname. UAexpert handles this with replacing the endpoint names with the public hostname.

```
2020-04-22T10:15:20.673Z        INFO    nodevalue/client.go:51  [OPCUA] Get all endpoints from opc.tcp://localhost:4840
2020-04-22T10:15:20.673Z        DEBUG   [Connect]       nodevalue/client.go:55  dial tcp 127.0.0.1:4840: connect: connection refused
2020-04-22T10:15:20.673Z        ERROR   nodevalue/client.go:60  [OPCUA] Failed to find suitable endpoint. Will try to switch to default [No security settings]. The following configurations are available for security0000000000000000:
2020-04-22T10:15:21.674Z        ERROR   nodevalue/nodevalue.go:153      [OPCUA] Tried to connect to OPC UA server 1 time(s). Without success.
2020-04-22T10:15:21.676Z        INFO    [monitoring]    log/log.go:154  Uptime: 1.039927422s
2020-04-22T10:15:21.676Z        INFO    [monitoring]    log/log.go:131  Stopping metrics logging.
2020-04-22T10:15:21.676Z        INFO    instance/beat.go:445    machinebeat stopped.
2020-04-22T10:15:21.676Z        ERROR   instance/beat.go:933    Exiting: 1 error: Connection was not possible
```

felix-lessoer, 2020-04-22:
I just made a test against this server and it works as expected. Based on your logs it looks like you have configured `opc.tcp://localhost:4840` to be the endpoint. This log entry will created directly from the config parameters.

This is the result in my log: 
```
2020-04-22T19:50:11.853+0200	INFO	nodevalue/client.go:51	[OPCUA] Get all endpoints from opc.tcp://opcua.rocks:4840/
2020-04-22T19:50:12.097+0200	INFO	nodevalue/client.go:65	[OPCUA] Policy URI: http://opcfoundation.org/UA/SecurityPolicy#None with security mode MessageSecurityModeNone
2020-04-22T19:50:12.097+0200	INFO	nodevalue/client.go:85	[OPCUA] Set to anonymous login
2020-04-22T19:50:12.334+0200	INFO	nodevalue/client.go:96	[OPCUA] Connection established
2020-04-22T19:50:12.334+0200	INFO	nodevalue/client.go:306	[OPCUA] Start browsing from node: root
2020-04-22T19:50:12.382+0200	INFO	nodevalue/client.go:339	Analyse node id i=85
2020-04-22T19:50:12.481+0200	INFO	nodevalue/client.go:339	Analyse node id i=2253
2020-04-22T19:50:12.575+0200	INFO	nodevalue/client.go:339	Analyse node id i=17634
2020-04-22T19:50:12.661+0200	INFO	nodevalue/client.go:339	Analyse node id i=2994
2020-04-22T19:50:12.661+0200	INFO	nodevalue/client.go:343	Add new node to subscription list: ID: i=2994| Type bool| Name Auditing
2020-04-22T19:50:12.661+0200	INFO	nodevalue/client.go:190	[OPCUA] Subscribe to node: i=2994
2020-04-22T19:50:12.750+0200	INFO	nodevalue/client.go:339	Analyse node id i=2255
2020-04-22T19:50:12.750+0200	INFO	nodevalue/client.go:343	Add new node to subscription list: ID: i=2255| Type string| Name NamespaceArray
2020-04-22T19:50:12.750+0200	INFO	nodevalue/client.go:190	[OPCUA] Subscribe to node: i=2255
2020-04-22T19:50:12.794+0200	INFO	nodevalue/client.go:242	[OPCUA] Created subscription with id 1
2020-04-22T19:50:12.835+0200	INFO	nodevalue/client.go:339	Analyse node id i=2254
2020-04-22T19:50:12.835+0200	INFO	nodevalue/client.go:343	Add new node to subscription list: ID: i=2254| Type string| Name ServerArray
2020-04-22T19:50:12.835+0200	INFO	nodevalue/client.go:190	[OPCUA] Subscribe to node: i=2254
2020-04-22T19:50:12.890+0200	INFO	nodevalue/client.go:242	[OPCUA] Created subscription with id 2
2020-04-22T19:50:12.927+0200	INFO	nodevalue/client.go:339	Analyse node id i=2267
2020-04-22T19:50:12.927+0200	INFO	nodevalue/client.go:343	Add new node to subscription list: ID: i=2267| Type byte| Name ServiceLevel
2020-04-22T19:50:12.927+0200	INFO	nodevalue/client.go:190	[OPCUA] Subscribe to node: i=2267
2020-04-22T19:50:12.976+0200	INFO	nodevalue/client.go:242	[OPCUA] Created subscription with id 3
2020-04-22T19:50:13.012+0200	INFO	nodevalue/client.go:339	Analyse node id i=12885
2020-04-22T19:50:13.012+0200	INFO	nodevalue/client.go:343	Add new node to subscription list: ID: i=12885| Type time.Time| Name EstimatedReturnTime
2020-04-22T19:50:13.012+0200	INFO	nodevalue/client.go:190	[OPCUA] Subscribe to node: i=12885
2020-04-22T19:50:13.057+0200	INFO	nodevalue/client.go:242	[OPCUA] Created subscription with id 4
2020-04-22T19:50:13.099+0200	INFO	nodevalue/client.go:339	Analyse node id i=11492
2020-04-22T19:50:13.151+0200	INFO	nodevalue/client.go:242	[OPCUA] Created subscription with id 5
2020-04-22T19:50:13.183+0200	INFO	nodevalue/client.go:339	Analyse node id i=11494
2020-04-22T19:50:13.268+0200	INFO	nodevalue/client.go:339	Analyse node id i=11493
2020-04-22T19:50:13.369+0200	INFO	nodevalue/client.go:339	Analyse node id ns=1;i=42
2020-04-22T19:50:13.455+0200	INFO	nodevalue/client.go:339	Analyse node id ns=1;i=62541
2020-04-22T19:50:13.548+0200	INFO	nodevalue/client.go:339	Analyse node id i=476450912
2020-04-22T19:50:13.634+0200	INFO	nodevalue/client.go:339	Analyse node id i=3751290663
2020-04-22T19:50:13.676+0200	INFO	cfgfile/reload.go:171	Config reloader started
2020-04-22T19:50:13.677+0200	WARN	[cfgwarn]	nodevalue/nodevalue.go:92	BETA: The OPCUA metricset is beta.
2020-04-22T19:50:13.677+0200	WARN	nodevalue/nodevalue.go:124	A new connection attempt was made. This gets ignored from this module
2020-04-22T19:50:13.677+0200	INFO	cfgfile/reload.go:226	Loading of config files completed.
2020-04-22T19:50:13.678+0200	INFO	nodevalue/nodevalue.go:193	[OPCUA] Publishing 5 new events
2020-04-22T19:50:14.678+0200	INFO	nodevalue/nodevalue.go:193	[OPCUA] Publishing 0 new events
```

cskowronnek, 2020-04-22:
This is indeed strange. I configured endpont:  "opc.tcp://opcua.rocks:4840/
localhost:4840 is also what UAexpert shows as the endpoint on a Windows PC. I also tried to resolve opcua.rocks and using the IP address I got the same behavior.

felix-lessoer, 2020-04-22:
Can you share your configuration? And are you sure that you only have 1 enabled configuration?

cskowronnek, 2020-04-25:
My bad. I had a typo and wrote endpont instead of endpoint in the config. It works with this server now, too.

cacamille3, 2020-05-06:
Hi,
I am from LOYTEC Electronics GmbH and responsible for the OPC UA Server on our device.
We have received the issue from our customer using MachineBeat.
I guess the issue could be solved onthe OPC UA Client side by removing the Timestamp value in the ViewDescription.

ViewDescription's timestamp (or viewVersion) value shall ONLY be present when you want to browse a precise View of an OPC UA Address Space.
Since you do not set a ViewDescription's viewId value, it does not make sense to set the timestamp value.

_A Wireshark capture from MachineBeat Go-OPCUA Client:_

 [image]

_The OPC UA Specification for the Browse Service:_
[image]

[image]

felix-lessoer, 2020-05-18:
I was able to reproduce the issue with the OPC UA server on Loytec LGATE-902 device.
I also found a solution to solve it in the underlying library. 

It will take some time to get that change into the library. For now I could send a pre-release that contains the fix. But it would be not possible to build the app on your own including that fix until it is merged into the lib.

Let me know how you would like to proceed.

felix-lessoer, 2020-06-23:
@cskowronnek I would like to close the ticket as this problem is solved now also for your OPC UA server. Are you okay with that?

closed 2020-06-29 by felix-lessoer