elastic/Machinebeat #14: Can't build Machinebeat from source

cskowronnek, 2020-04-15:
First I changed beats version in go.mod from v7.6.1 to v7.6.2+incompatible because it couldn't find 7.6.1
I also had to create a "v7" folder under /src.../beats and copy the libbeat folder into it because it couldn't find the sources.
I'm now stuck here (btw: what to enter for module name and metricset?)

```
vendor/github.com/elastic/beats/metricbeat/Makefile:84: Warnung: Die Befehle für das Ziel „integration-tests“ werden überschrieben
vendor/github.com/elastic/beats/libbeat/scripts/Makefile:216: Warnung: Alte Befehle für das Ziel „integration-tests“ werden ignoriert
mkdir -p vendor/github.com/elastic
cp -R /src/github.com/elastic/beats vendor/github.com/elastic/
ln -sf /home/pi/Machinebeat/vendor/github.com/elastic/beats/metricbeat/scripts/generate_imports_helper.py /home/pi/Machinebeat/vendor/github.com/elastic/beats/script/generate_imports_helper.py
rm -rf vendor/github.com/elastic/beats/.git vendor/github.com/elastic/beats/x-pack
mkdir -p vendor/github.com/magefile
cp -R /src/github.com/elastic/beats/vendor/github.com/magefile/mage vendor/github.com/magefile
git init
Bestehendes Git-Repository in /home/pi/Machinebeat/.git/ neuinitialisiert
#call make recursively so we can reload the above include.
#Only needed during the first setup phase, before /vendor exists
make create-metricset collect git-add
make[1]: Verzeichnis „/home/pi/Machinebeat“ wird betreten
vendor/github.com/elastic/beats/metricbeat/Makefile:84: Warnung: Die Befehle für das Ziel „integration-tests“ werden überschrieben
vendor/github.com/elastic/beats/libbeat/scripts/Makefile:216: Warnung: Alte Befehle für das Ziel „integration-tests“ werden ignoriert
Module name:
Metricset name:
Module already exists. Skipping creating module
Metricset already exists. Skipping creating metricset
go run -mod=vendor ./vendor/github.com/elastic/beats/metricbeat/scripts/assets/assets.go ./vendor/github.com/elastic/beats/metricbeat/module
build github.com/elastic/beats/metricbeat/scripts/assets: cannot find module for path github.com/pkg/errors
make[1]: *** [vendor/github.com/elastic/beats/metricbeat/Makefile:77: assets] Fehler 1
make[1]: Verzeichnis „/home/pi/Machinebeat“ wird verlassen
make: *** [Makefile:19: setup] Fehler 2
```

felix-lessoer, 2020-04-15:
You don't need to create a new metricset . So make create-metricset is not necessary.
If you would like to build it, just go into the machinebeat folder and use `go build` .. You may need to load the dependencies. Thats enough for now.

felix-lessoer, 2020-04-25:
Can we close this one?

cskowronnek, 2020-04-25:
Yes, I thin so.

closed 2020-04-25 by cskowronnek