elastic/Machinebeat #15: ARM build

cskowronnek, 2020-04-24:
We would like to run Machinebeat on a Raspberry Pi 4. Is it possible to cross compile for ARM8?

felix-lessoer, 2020-04-24:
Just build it for ARM8. Could you check that it is working for you?
https://ela.st/machinebeat-arm

cskowronnek, 2020-04-25:
Unfortunately our RPi 4 is still running on a 32-bit kernel as the 64-bit kernel is still experimental. Could you build a 32-bit ARM version as well? 
```
:~/machinebeat $ ./machinebeat -c machinebeat.yml -e 
bash: ./machinebeat: Kann die Binärdatei nicht ausführen: Fehler im Format der Programmdatei
:~/machinebeat $ uname -a
Linux pit-dashboard 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux
:~/machinebeat $ file machinebeat
machinebeat: ELF 64-bit LSB executable, ARM aarch64, version 1 (SYSV), statically linked, Go BuildID=ZkCFiY1nlW9HGcMH_dyD/Ucjuk0MTg9dHCVbUyaTj/EvWSMw-4PdQOtQjiuvqS/YTz_cwizH3r5efmVxXVP, not stripped
```

felix-lessoer, 2020-04-25:
I used ARM8 based in this documentation:
https://github.com/golang/go/wiki/GoArm

ARM7 seems to be for 32bit

felix-lessoer, 2020-04-25:
Same link but it contains the arm7 version now.

cskowronnek, 2020-04-25:
Thanks! The binary runs. I'll ship it to our test device and see if it runs stable

cskowronnek, 2020-04-28:
The build runs stable on two Raspberry Pi 4 until now.

felix-lessoer, 2020-04-28:
Sounds like we can close that issue :)

cskowronnek, 2020-04-29:
Yes, we can. Could you please build future version also for ARM?

closed 2020-04-29 by cskowronnek