elastic/Machinebeat #16: make setup does not seem to work

Mpdreamz, 2020-06-18:
When it tries to run `create-metricset` it calls mage in the wrong location resulting in:

```
2020/06/18 15:55:32 Found Elastic Beats dir at /home/mpdreamz/projects/go/src/github.com/elastic/Machinebeat/vendor/github.com/elastic/beats
Unknown target specified: createMetricset
```

If I update the make call so that the pwd is scoped to metricbeat.  

```
--- a/Makefile
+++ b/Makefile
@@ -16,7 +16,7 @@ MAGE_IMPORT_PATH=${BEAT_PATH}/vendor/github.com/magefile/mage
 setup: copy-vendor git-init
        #call make recursively so we can reload the above include.
        #Only needed during the first setup phase, before /vendor exists
-       $(MAKE) create-metricset collect git-add
+       $(MAKE) -C $(ES_BEATS)/metricbeat create-metricset collect git-add
```

It will succeed to call `mage createMetricet` but then none of the file references seem to resolve:

```
make -C ./vendor/github.com/elastic/beats/metricbeat create-metricset collect git-add
make[1]: Entering directory '/home/mpdreamz/projects/go/src/github.com/elastic/Machinebeat/vendor/github.com/elastic/beats/metricbeat'
mage createMetricset
ERROR: Could not open requirements file: [Errno 2] No such file or directory: '/home/mpdreamz/projects/go/src/github.com/elastic/Machinebeat/libbeat/tests/system/requirements.txt'
Error: running "/home/mpdreamz/projects/go/src/github.com/elastic/Machinebeat/build/ve/linux/bin/pip install --quiet -Ur /home/mpdreamz/projects/go/src/github.com/elastic/Machinebeat/libbeat/tests/system/requirements.txt" failed with exit code 1
```

`make --version`

```
GNU Make 4.2.1
Built for x86_64-pc-linux-gnu
Copyright (C) 1988-2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

`mage --version`

```
Mage Build Tool v1.9.0-12-g310e198
Build Date: 2020-06-18T14:34:58+02:00
Commit: 310e198
built with: go1.14.4
```

`go --version`

```
go version go1.14.4 linux/amd64
```

felix-lessoer, 2020-06-18:
I have disabled the make components in the current version as the beat framework is changing from v7.6 to v7.7.
But its enough to run "go build" if you would like to build it on your own. You can also use the prebuild binaries.
The linux is build is made for amd64

felix-lessoer, 2020-06-22:
Can you work with the pre build version? 
If yes it would be nice if you could close the ticket.

closed 2020-06-29 by felix-lessoer