elastic/Machinebeat #17: Multiple endpoints - connection issues and result data errors

MagnusArias, 2020-06-25:
I need to monitor some nodes from bunch of different endpoints. The issue arises when more than one endpoint is configured. When multiple endpoint are configured in `opcua.yml`, it is high chance to connect only to first endpoint. Also, there is no information in log file, that we are connected to another endpoint (only the first one).

Here is the `opcua.yml` config file.

```yaml
# ----
# Endpoint 1
# ----
- module: opcua
  metricsets: ["nodevalue"]
  enabled: true
  period: 1s

  endpoint: "opc.tcp://<endpoint_IP_1:Port>"

  policy: "None"
  mode: "None"

  subscribe: true
  browse:
  - enabled: true
    maxLevel: 3
    maxNodePerParent: 5

  retryOnError: 5
  maxThreads: 50

  nodes:
  - id: "ns=3;s=Lebensbit[0]"
    label: "Lbit EP1"
  - id: "ns=3;s=Tag[0]"
    label: "Tag EP1"

# ----
# Endpoint 2
# ----
- module: opcua
  metricsets: ["nodevalue"]
  enabled: true
  period: 1s

  endpoint: "opc.tcp://<endpoint_IP_2:Port>"
  policy: "None"
  mode: "None"

  subscribe: true
  browse:
  - enabled: true
    maxLevel: 3
    maxNodePerParent: 5

  retryOnError: 5
  maxThreads: 50

  nodes:
  - id: "ns=3;s=Lebensbit[0]"
    label: "Lbit EP2"
  - id: "ns=3;s=Tag[0]"
    label: "Tag EP2"
```

Also, if those endpoints are connected, **node labels** on other endpoints are overriden by node labels from first endpoint.

Below is partial JSON data, first from EP1 (correct), second from EP2 (wrong)

```json
{
  "opcua": {
    "nodevalue": {
      "state": "OK",
      "created": "2020-06-25 06:03:15.0713383 +0000 UTC",
      "uint32": 1
    },
    "node": {
      "Label": "Lbit EP1",
      "Name": "1000 Lebensbit",
      "DataType": "uint32",
      "ID": "ns=3;s=Lebensbit[0]"
    },
    "endpoint": "opc.tcp://<IP_Endpoint_1:Port>"
  }
}
```

```json
{
  "opcua": {
    "nodevalue": {
      "created": "2020-06-25 06:03:16.0872965 +0000 UTC",
      "uint32": 1,
      "state": "OK"
    },
    "node": {
      "Name": "1000 Lebensbit",
      "DataType": "uint32",
      "ID": "ns=3;s=Lebensbit[0]",
      "Label": "LBit EP1"
    },
    "endpoint": "opc.tcp://<IP_Endpoint_2:Port>"
  }
}
```

If I'm wrong, please tell me, how should look the correct configuration file for multiple endpoint.

felix-lessoer, 2020-06-29:
Hey @MagnusArias ,

I'm not sure how quick I can fix this problem. The two instances should not influence each other. But based on your data it looks like they do.
Would it be an option for you to run two separate processes of Machinebeat in the meanwhile?

MagnusArias, 2020-06-29:
Hey @felix-lessoer ,

I think this will be possible to run different instances of Machinebeat, but in the future I will need to monitor about 30+ different endpoints, gathering at least 5 nodes. The only thing I can change is period, it can be expanded to 5 seconds.

In my opinion, those 30 instances will kill my machine, where I already have a lot of other modules for ES.

felix-lessoer, 2020-10-23:
Hi @MagnusArias ,

I fixed the problem now and released a new version.
It is now possible to run independent opcua modules in the same machinebeat.

Would be nice if you could test it and provide feedback.
If everything works feel free to close the issue, pls.

closed 2020-10-31 by felix-lessoer