elastic/Machinebeat #18: Panic while starting machinebeat

bdauvissat, 2020-06-29:
Hi !

I'm testing Machinebeat with an OPC Server emulator : [https://www.prosysopc.com/products/opc-ua-simulation-server/](url)

I'm just created a user "bdt" on this server.

Then, I updated the opcua.yml file with the URL, the user and the password.

Finally, I started the machinebeat.exe and had the following logs : 
`
2020-06-29T17:20:58.951+0200	INFO	instance/beat.go:622	Home path: [D:\Projets\Elasticsearch\opc\machinebeat] Config path: [D:\Projets\Elasticsearch\opc\machinebeat] Data path: [D:\Projets\Elasticsearch\opc\machinebeat\data] Logs path: [D:\Projets\Elasticsearch\opc\machinebeat\logs]
2020-06-29T17:20:58.986+0200	INFO	instance/beat.go:630	Beat ID: 6b8a6085-8c43-4bc0-9a0b-458778cb2ff0
2020-06-29T17:20:59.069+0200	INFO	[beat]	instance/beat.go:958	Beat info	{"system_info": {"beat": {"path": {"config": "D:\\Projets\\Elasticsearch\\opc\\machinebeat", "data": "D:\\Projets\\Elasticsearch\\opc\\machinebeat\\data", "home": "D:\\Projets\\Elasticsearch\\opc\\machinebeat", "logs": "D:\\Projets\\Elasticsearch\\opc\\machinebeat\\logs"}, "type": "machinebeat", "uuid": "6b8a6085-8c43-4bc0-9a0b-458778cb2ff0"}}}
2020-06-29T17:20:59.070+0200	INFO	[beat]	instance/beat.go:967	Build info	{"system_info": {"build": {"commit": "unknown", "libbeat": "7.6.1", "time": "1754-08-30T22:43:41.128Z", "version": "7.6.1"}}}
2020-06-29T17:20:59.070+0200	INFO	[beat]	instance/beat.go:970	Go runtime info	{"system_info": {"go": {"os":"windows","arch":"amd64","max_procs":8,"version":"go1.12.4"}}}
2020-06-29T17:20:59.070+0200	INFO	add_cloud_metadata/add_cloud_metadata.go:89	add_cloud_metadata: hosting provider type not detected.
2020-06-29T17:20:59.151+0200	INFO	[beat]	instance/beat.go:974	Host info	{"system_info": {"host": {"architecture":"x86_64","boot_time":"2020-06-28T18:25:01.66+02:00","name":"BDT-10","ip":["fe80::c8f9:901b:268:4fbd/64","169.254.79.189/16","fe80::5c6f:1f32:38e6:607e/64","192.168.56.1/24","fe80::2cb5:c6ee:7f33:a1b7/64","169.254.161.183/16","fe80::3c67:74ce:ed22:e856/64","169.254.232.86/16","fe80::c930:cdf5:1426:1f82/64","192.168.125.10/30","fe80::751c:f339:36cd:f708/64","169.254.247.8/16","fe80::959d:f53c:7924:95f0/64","192.168.229.1/24","fe80::9c8a:74e7:1859:7a48/64","192.168.211.1/24","fe80::d520:6ad8:7d59:91/64","192.168.0.24/24","::1/128","127.0.0.1/8"],"kernel_version":"10.0.18362.900 (WinBuild.160101.0800)","mac":["e4:e7:49:43:ce:e1","0a:00:27:00:00:18","1c:1b:b5:10:94:5c","1e:1b:b5:10:94:5b","00:ff:60:49:f5:80","00:ff:6e:80:17:95","00:50:56:c0:00:01","00:50:56:c0:00:08","1c:1b:b5:10:94:5b"],"os":{"family":"windows","platform":"windows","name":"Windows 10 Pro","version":"10.0","major":10,"minor":0,"patch":0,"build":"18363.900"},"timezone":"CEST","timezone_offset_sec":7200,"id":"cd71d502-a2d2-48ca-9c2a-d9adb85e1121"}}}
2020-06-29T17:20:59.154+0200	INFO	[beat]	instance/beat.go:1003	Process info	{"system_info": {"process": {"cwd": "D:\\Projets\\Elasticsearch\\opc\\machinebeat", "exe": "D:\\Projets\\Elasticsearch\\opc\\machinebeat\\machinebeat.exe", "name": "machinebeat.exe", "pid": 19436, "ppid": 19188, "start_time": "2020-06-29T17:20:57.774+0200"}}}
2020-06-29T17:20:59.154+0200	INFO	instance/beat.go:298	Setup Beat: machinebeat; Version: 7.6.1
2020-06-29T17:20:59.154+0200	INFO	[index-management]	idxmgmt/std.go:182	Set output.elasticsearch.index to 'machinebeat-7.6.1' as ILM is enabled.
2020-06-29T17:20:59.154+0200	INFO	elasticsearch/client.go:174	Elasticsearch url: http://localhost:9200
2020-06-29T17:20:59.155+0200	INFO	[publisher]	pipeline/module.go:110	Beat name: BDT-10
2020-06-29T17:20:59.155+0200	INFO	instance/beat.go:439	machinebeat start running.
2020-06-29T17:20:59.155+0200	INFO	[monitoring]	log/log.go:118	Starting metrics logging every 30s
2020-06-29T17:20:59.155+0200	WARN	[cfgwarn]	nodevalue/nodevalue.go:92	BETA: The OPCUA metricset is beta.
2020-06-29T17:20:59.155+0200	INFO	nodevalue/client.go:54	[OPCUA] Get all endpoints from opc.tcp://BDT-10.GROUPE-SPC.local:53530/OPCUA/SimulationServer
2020-06-29T17:20:59.167+0200	INFO	nodevalue/client.go:68	[OPCUA] Policy URI: http://opcfoundation.org/UA/SecurityPolicy#None with security mode MessageSecurityModeNone
2020-06-29T17:20:59.167+0200	INFO	nodevalue/client.go:84	[OPCUA] Set authentication information
2020-06-29T17:20:59.167+0200	INFO	nodevalue/client.go:85	[OPCUA] User: bdt
2020-06-29T17:20:59.321+0200	INFO	nodevalue/client.go:99	[OPCUA] Connection established
2020-06-29T17:20:59.321+0200	INFO	nodevalue/nodevalue.go:132	Browsing is enabled. Data collection will start after discovery. Based on your server and browsing configuration this can take some time.
2020-06-29T17:20:59.322+0200	INFO	nodevalue/client.go:304	[OPCUA] Start browsing from Objects and Views folder
2020-06-29T17:20:59.322+0200	INFO	nodevalue/client.go:349	Analyse node id i=85
2020-06-29T17:20:59.345+0200	INFO	nodevalue/client.go:349	Analyse node id i=2253
2020-06-29T17:20:59.349+0200	INFO	[monitoring]	log/log.go:153	Total non-zero metrics	{"monitoring": {"metrics": {"beat":{"cpu":{"system":{"ticks":109,"time":{"ms":109}},"total":{"ticks":171,"time":{"ms":171},"value":171},"user":{"ticks":62,"time":{"ms":62}}},"handles":{"open":263},"info":{"ephemeral_id":"5f59b0df-151f-4ad9-8045-3f993b15eeb6","uptime":{"ms":1548}},"memstats":{"gc_next":5337456,"memory_alloc":4127496,"memory_total":8028656,"rss":35614720},"runtime":{"goroutines":18}},"libbeat":{"config":{"module":{"running":0}},"output":{"type":"elasticsearch"},"pipeline":{"clients":0,"events":{"active":0}}},"system":{"cpu":{"cores":8}}}}}
2020-06-29T17:20:59.349+0200	INFO	[monitoring]	log/log.go:154	Uptime: 1.5492895s
2020-06-29T17:20:59.349+0200	INFO	[monitoring]	log/log.go:131	Stopping metrics logging.
2020-06-29T17:20:59.349+0200	INFO	runtime/panic.go:522	machinebeat stopped.
2020-06-29T17:20:59.394+0200	FATAL	[machinebeat]	instance/beat.go:163	Failed due to panic.	{"panic": "runtime error: index out of range", "stack": "github.com/elastic/beats/libbeat/cmd/instance.Run.func1.1\n\tC:/Users/felix/projekte/machinebeat/src/github.com/elastic/beats/libbeat/cmd/instance/beat.go:164\nruntime.gopanic\n\tC:/Go/src/runtime/panic.go:522\nruntime.panicindex\n\tC:/Go/src/runtime/panic.go:44\ngithub.com/elastic/machinebeat/module/opcua/nodevalue.browse\n\tC:/Users/felix/projekte/machinebeat/src/github.com/elastic/machinebeat/module/opcua/nodevalue/client.go:359\ngithub.com/elastic/machinebeat/module/opcua/nodevalue.browse\n\tC:/Users/felix/projekte/machinebeat/src/github.com/elastic/machinebeat/module/opcua/nodevalue/client.go:369\ngithub.com/elastic/machinebeat/module/opcua/nodevalue.startBrowse\n\tC:/Users/felix/projekte/machinebeat/src/github.com/elastic/machinebeat/module/opcua/nodevalue/client.go:316\ngithub.com/elastic/machinebeat/module/opcua/nodevalue.New\n\tC:/Users/felix/projekte/machinebeat/src/github.com/elastic/machinebeat/module/opcua/nodevalue/nodevalue.go:135\ngithub.com/elastic/beats/metricbeat/mb.initMetricSets\n\tC:/Users/felix/projekte/machinebeat/src/github.com/elastic/beats/metricbeat/mb/builders.go:139\ngithub.com/elastic/beats/metricbeat/mb.NewModule\n\tC:/Users/felix/projekte/machinebeat/src/github.com/elastic/beats/metricbeat/mb/builders.go:65\ngithub.com/elastic/beats/metricbeat/mb/module.NewWrapper\n\tC:/Users/felix/projekte/machinebeat/src/github.com/elastic/beats/metricbeat/mb/module/wrapper.go:84\ngithub.com/elastic/beats/metricbeat/mb/module.(*Factory).CheckConfig\n\tC:/Users/felix/projekte/machinebeat/src/github.com/elastic/beats/metricbeat/mb/module/factory.go:72\ngithub.com/elastic/beats/libbeat/cfgfile.(*Reloader).Check\n\tC:/Users/felix/projekte/machinebeat/src/github.com/elastic/beats/libbeat/cfgfile/reload.go:157\ngithub.com/elastic/beats/metricbeat/beater.(*Metricbeat).Run\n\tC:/Users/felix/projekte/machinebeat/src/github.com/elastic/beats/metricbeat/beater/metricbeat.go:255\ngithub.com/elastic/beats/libbeat/cmd/instance.(*Beat).launch\n\tC:/Users/felix/projekte/machinebeat/src/github.com/elastic/beats/libbeat/cmd/instance/beat.go:445\ngithub.com/elastic/beats/libbeat/cmd/instance.Run.func1\n\tC:/Users/felix/projekte/machinebeat/src/github.com/elastic/beats/libbeat/cmd/instance/beat.go:188\ngithub.com/elastic/beats/libbeat/cmd/instance.Run\n\tC:/Users/felix/projekte/machinebeat/src/github.com/elastic/beats/libbeat/cmd/instance/beat.go:189\nmain.main\n\tC:/Users/felix/projekte/machinebeat/src/github.com/elastic/machinebeat/main.go:21\nruntime.main\n\tC:/Go/src/runtime/proc.go:200"}
`
Can you tell me what went wrong ?

Thanks.

felix-lessoer, 2020-07-01:
It hasn't found any node to collect data from.
In the next version I will prevent it from panicking if this happens.

bdauvissat, 2020-07-01:
Hi.
Here are my conf file, a screenshot of the OPC UA Server and the log file.
I don't understand why I disabled browsing and the log says that browsing is still enabled.
Regards
[machinebeat.log](https://github.com/elastic/Machinebeat/files/4859400/machinebeat.log)
[image]

[image]

felix-lessoer, 2020-07-01:
Try `browse.enabled: false`
This should disable browsing and will take the result directly from the configured node.

bdauvissat, 2020-07-01:
It works.

But why isn't the browsing mode working so I can work with several endpoints ?

felix-lessoer, 2020-07-02:
You can work with multiple nodes also without browsing. Just add another one to your config.

The browsing works for many different implementations of OPC UA servers. But we have seen that it not works for all.

To provide more details it would be good to see the debug logs.

I will also test it with your configuration when I find time.

Thanks for your contribution.

bdauvissat, 2020-07-02:
Thanks for your help.

closed 2020-08-27 by felix-lessoer