elastic/Machinebeat #19: How to build machine beat from source

arunnalpet, 2020-07-17:
Machinebeat is failing to connect to OPCUA server. I may need to tweak TCP parameters in GO client. 
Where can I find build instructions, so that I can tweak the parameters and rebuild machinebeat?

felix-lessoer, 2020-07-17:
Can you show the logs? You can tweak everything needed using the config.

arunnalpet, 2020-07-20:
Hi, 
In a different experiment where i was using Free-pythonOPCUA client, I had to tweak the messagesize and set it to '65535' for establishing a successful connection. (Default was 0, i.e. No limit)

I believe Machine beat is using GO client under the hood. I was looking for similar options to tweak these connectivity parameters. 
Here are the logs of machine beat. 

```
2020-07-20T13:49:51.739+0530    INFO    instance/beat.go:622    Home path: [c:\ELK\machinebeat] Config path: [c:\ELK\machinebeat] Data path: [c:\ELK\machinebeat\data] Logs path: [c:\ELK\machinebeat\logs]
2020-07-20T13:49:51.742+0530    INFO    instance/beat.go:630    Beat ID: c6dcedad-0963-4627-909a-ac1e49824860
2020-07-20T13:49:51.866+0530    INFO    [beat]  instance/beat.go:958    Beat info       {"system_info": {"beat": {"path": {"config": "c:\\ELK\\machinebeat", "data": "c:\\ELK\\machinebeat\\data", "home": "c:\\ELK\\machinebeat", "logs": "c:\\ELK\\machinebeat\\logs"}, "type": "machinebeat", "uuid": "c6dcedad-0963-4627-909a-ac1e49824860"}}}
2020-07-20T13:49:51.867+0530    INFO    [beat]  instance/beat.go:967    Build info      {"system_info": {"build": {"commit": "unknown", "libbeat": "7.6.1", "time": "1754-08-30T22:43:41.128Z", "version": "7.6.1"}}}
2020-07-20T13:49:51.868+0530    INFO    [beat]  instance/beat.go:970    Go runtime info {"system_info": {"go": {"os":"windows","arch":"amd64","max_procs":8,"version":"go1.12.4"}}}
2020-07-20T13:49:51.867+0530    INFO    add_cloud_metadata/add_cloud_metadata.go:89     add_cloud_metadata: hosting provider type not detected.
2020-07-20T13:49:51.982+0530    INFO    [beat]  instance/beat.go:974    Host info       {"system_info": {"host": {"architecture":"x86_64","boot_time":"2020-07-13T18:57:49.52+05:30","name":"FLDJBFW2QC","ip":["fe80::7906:91c1:3ac4:d4e6/64","16.242.165.23/32","fe80::34d0:143:3fdc:ade3/64","169.254.173.227/16","fe80::754d:33d0:58ec:b674/64","169.254.182.116/16","fe80::88c:3533:cafd:d029/64","10.188.254.17/24","fe80::c18c:bc05:7d48:6670/64","169.254.102.112/16","fe80::6dda:6216:ef4b:d12e/64","192.168.0.106/24","fe80::4135:ebc5:b2bb:d215/64","169.254.210.21/16","::1/128","127.0.0.1/8"],"kernel_version":"10.0.16299.1932 (WinBuild.160101.0800)","mac":["02:05:85:7f:eb:80","c4:65:16:84:60:0f","c4:65:16:84:60:11","00:ff:85:3f:61:6d","d8:f2:ca:b2:71:91","d8:f2:ca:b2:71:90","d8:f2:ca:b2:71:94"],"os":{"family":"windows","platform":"windows","name":"Windows 10 Enterprise","version":"10.0","major":10,"minor":0,"patch":0,"build":"16299.1932"},"timezone":"IST","timezone_offset_sec":19800,"id":"130a4d12-bc5f-43bb-ad56-431f5d783231"}}}
2020-07-20T13:49:51.985+0530    INFO    [beat]  instance/beat.go:1003   Process info    {"system_info": {"process": {"cwd": "c:\\ELK\\machinebeat", "exe": "C:\\ELK\\machinebeat\\machinebeat.exe", "name": "machinebeat.exe", "pid": 594332, "ppid": 510724, "start_time": "2020-07-20T13:49:47.443+0530"}}}
2020-07-20T13:49:51.985+0530    INFO    instance/beat.go:298    Setup Beat: machinebeat; Version: 7.6.1
2020-07-20T13:49:51.986+0530    INFO    [index-management]      idxmgmt/std.go:182      Set output.elasticsearch.index to 'machinebeat-7.6.1' as ILM is enabled.
2020-07-20T13:49:51.986+0530    INFO    elasticsearch/client.go:174     Elasticsearch url: http://localhost:9200
2020-07-20T13:49:51.986+0530    INFO    [publisher]     pipeline/module.go:110  Beat name: FLDJBFW2QC
2020-07-20T13:49:51.987+0530    INFO    instance/beat.go:439    machinebeat start running.
2020-07-20T13:49:51.987+0530    INFO    [monitoring]    log/log.go:118  Starting metrics logging every 30s
2020-07-20T13:49:51.990+0530    WARN    [cfgwarn]       nodevalue/nodevalue.go:92       BETA: The OPCUA metricset is beta.
2020-07-20T13:49:51.990+0530    INFO    nodevalue/client.go:54  [OPCUA] Get all endpoints from opc.tcp://x.x.x.x:4840
2020-07-20T13:49:52.615+0530    ERROR   nodevalue/client.go:63  [OPCUA] Failed to find suitable endpoint. Will try to switch to default [No security settings]. The following configurations are available for security:
2020-07-20T13:49:52.615+0530    INFO    nodevalue/client.go:39  [OPCUA] This server has no endpoints. This can happen when the OPC UA server can't be reached. Are you sure that the endpoint is right?
2020-07-20T13:49:54.250+0530    ERROR   nodevalue/nodevalue.go:165      [OPCUA] Tried to connect to OPC UA server 1 time(s). Without success.
2020-07-20T13:49:54.251+0530    INFO    [monitoring]    log/log.go:153  Total non-zero metrics  {"monitoring": {"metrics": {"beat":{"cpu":{"system":{"ticks":265,"time":{"ms":265}},"total":{"ticks":327,"time":{"ms":327},"value":0},"user":{"ticks":62,"time":{"ms":62}}},"handles":{"open":244},"info":{"ephemeral_id":"339fd5d2-0b83-4e2b-a8a2-983c325da5ca","uptime":{"ms":6503}},"memstats":{"gc_next":5312528,"memory_alloc":3235664,"memory_total":6764120,"rss":34926592},"runtime":{"goroutines":17}},"libbeat":{"config":{"module":{"running":0}},"output":{"type":"elasticsearch"},"pipeline":{"clients":0,"events":{"active":0}}},"system":{"cpu":{"cores":8}}}}}
2020-07-20T13:49:54.252+0530    INFO    [monitoring]    log/log.go:154  Uptime: 6.505015s
2020-07-20T13:49:54.252+0530    INFO    [monitoring]    log/log.go:131  Stopping metrics logging.
2020-07-20T13:49:54.253+0530    INFO    instance/beat.go:445    machinebeat stopped.
2020-07-20T13:49:54.257+0530    ERROR   instance/beat.go:933    Exiting: 1 error: Connection was not possible
Exiting: 1 error: Connection was not possible
```

[... cut ...]
test_find_servers (__main__.Tests) ... ok
test_get_namespace_array (__main__.Tests) ... ok
test_get_root (__main__.Tests) ... ok
test_get_root_children (__main__.Tests) ... ok
test_get_server_node (__main__.Tests) ... ok
test_subscribe_server_time (__main__.Tests) ... Timedelta is  44029 days, 16:26:06.457659
Received data for unknown subscription: 0
Received data for unknown subscription: 0
ok

----------------------------------------------------------------------
Ran 9 tests in 17.209s

OK
```

felix-lessoer, 2020-08-18:
I fixed the build process, so you should be able to build it on your own using the latest version in the repo.

closed 2020-08-27 by felix-lessoer