elastic/Machinebeat #20: Issue with Node Configurations in OPCUA.yml

CompsAndData, 2020-08-10:
Hello,

I am attempting to collect data from specific nodes from a Kepware OPC server.  I have added specific tags to the node configuration section and I have tried to turn browsing off.  Attached is the opcua.yml config file and some debug info.  When I try to start machinebeat, it appears that it tries to browse anyway and other information is being sent to Elastic.

Thanks for any help!

[log.txt](https://github.com/elastic/Machinebeat/files/5052952/log.txt)

[opcua.txt](https://github.com/elastic/Machinebeat/files/5052940/opcua.txt)

felix-lessoer, 2020-08-11:
You need to set 
`browse.enabled: false`
The other notion is not working anymore.

CompsAndData, 2020-08-12:
Thanks so much that worked perfect!

closed 2020-08-12 by CompsAndData