elastic/Machinebeat #21: Regarding anonymous login issue

dongwoo-john-ku, 2020-08-11:
Hello,

I'm writing to inquire about 'anonymous login' issue.
First of all, thank you for making machinebeat.

It looks a nice tool and I succeed to connect to the demo endpoint using your program without any problem.
 * demo endpoint: opc.tcp://opcuaserver.com:48010

However, I have a problem when I connect to my OPC UA server.
My local endpoint information like below and I checked the OPC UA server has no problem with another OPC/UA client tools 
 * local endpoint: opc.tcp://host.docker.internal:53530/OPCUA/SimulationServer

Is there some configuration what I can do?
Here is disconnection full log information and thank you in advanced. 

`2020-08-11T17:44:35.246+0900    INFO    instance/beat.go:622    Home path: [C:\Users\dwku\OneDrive\바탕 화면\machinebeat] Config path: [C:\Users\dwku\OneDrive\바탕 화면\machinebeat] Data path: [C:\Users\dwku\OneDrive\바탕 화면\machinebeat\data] Logs path: [C:\Users\dwku\OneDrive\바탕 화면\machinebeat\logs]
2020-08-11T17:44:35.307+0900    INFO    instance/beat.go:630    Beat ID: d51d9d17-d145-489f-b43d-f9953bc4d771
2020-08-11T17:44:35.355+0900    INFO    [beat]  instance/beat.go:958    Beat info       {"system_info": {"beat": {"path": {"config": "C:\\Users\\dwku\\OneDrive\\바탕 화면\\machinebeat", "data": "C:\\Users\\dwku\\OneDrive\\바탕 화면\\machinebeat\\data", "home": "C:\\Users\\dwku\\OneDrive\\바탕 화면\\machinebeat", "logs": "C:\\Users\\dwku\\OneDrive\\바탕 화면\\machinebeat\\logs"}, "type": "machinebeat", "uuid": "d51d9d17-d145-489f-b43d-f9953bc4d771"}}}
2020-08-11T17:44:35.356+0900    INFO    [beat]  instance/beat.go:967    Build info      {"system_info": {"build": {"commit": "unknown", "libbeat": "7.6.1", "time": "1754-08-30T22:43:41.128Z", "version": "7.6.1"}}}
2020-08-11T17:44:35.356+0900    INFO    [beat]  instance/beat.go:970    Go runtime info {"system_info": {"go": {"os":"windows","arch":"amd64","max_procs":8,"version":"go1.12.4"}}}
2020-08-11T17:44:35.358+0900    INFO    add_cloud_metadata/add_cloud_metadata.go:89     add_cloud_metadata: hosting provider type not detected.
2020-08-11T17:44:35.402+0900    INFO    [beat]  instance/beat.go:974    Host info       {"system_info": {"host": {"architecture":"x86_64","boot_time":"2020-08-11T13:44:01.31+09:00","name":"SEOUL-DKU","ip":["fe80::cc0f:37ad:34f1:3009/64","169.254.48.9/16","fe80::90e8:fdde:a2e0:8875/64","169.254.136.117/16","fe80::49f8:3301:4b72:98ad/64","192.168.0.6/24","fe80::a4fd:5499:5ae7:de0a/64","169.254.222.10/16","::1/128","127.0.0.1/8","fe80::4928:1421:1cf2:2d4b/64","172.25.129.17/28"],"kernel_version":"10.0.17134.1610 (WinBuild.160101.0800)","mac":["92:32:4b:1c:90:47","d2:32:4b:1c:90:47","90:32:4b:1c:90:47","90:32:4b:1c:90:48","72:15:91:d4:83:ca"],"os":{"family":"windows","platform":"windows","name":"Windows 10 Enterprise","version":"10.0","major":10,"minor":0,"patch":0,"build":"17134.1610"},"timezone":"KST","timezone_offset_sec":32400,"id":"e47d409c-3cca-409f-a573-66904f0b01e3"}}}
2020-08-11T17:44:35.405+0900    INFO    [beat]  instance/beat.go:1003   Process info    {"system_info": {"process": {"cwd": "C:\\Users\\dwku\\OneDrive\\바탕  화면\\machinebeat", "exe": "C:\\Users\\dwku\\OneDrive\\\ufffd\ufffd\ufffd\ufffd ȭ\ufffd\ufffd\\machinebeat\\machinebeat.exe", "name": "machinebeat.exe", "pid": 22124, "ppid": 22836, "start_time": "2020-08-11T17:44:34.360+0900"}}}
2020-08-11T17:44:35.406+0900    INFO    instance/beat.go:298    Setup Beat: machinebeat; Version: 7.6.1
2020-08-11T17:44:35.406+0900    INFO    [index-management]      idxmgmt/std.go:182      Set output.elasticsearch.index to 'machinebeat-7.6.1' as ILM is enabled.
2020-08-11T17:44:35.407+0900    INFO    elasticsearch/client.go:174     Elasticsearch url: http://localhost:9200
2020-08-11T17:44:35.407+0900    INFO    [publisher]     pipeline/module.go:110  Beat name: SEOUL-DKU
2020-08-11T17:44:35.408+0900    INFO    instance/beat.go:439    machinebeat start running.
2020-08-11T17:44:35.408+0900    INFO    [monitoring]    log/log.go:118  Starting metrics logging every 30s
2020-08-11T17:44:35.410+0900    WARN    [cfgwarn]       nodevalue/nodevalue.go:92       BETA: The OPCUA metricset is beta.
2020-08-11T17:44:35.410+0900    INFO    nodevalue/client.go:54  [OPCUA] Get all endpoints from opc.tcp://host.docker.internal:53530/OPCUA/SimulationServer
2020-08-11T17:44:35.432+0900    INFO    nodevalue/client.go:68  [OPCUA] Policy URI: http://opcfoundation.org/UA/SecurityPolicy#None with security mode MessageSecurityModeNone
2020-08-11T17:44:35.432+0900    INFO    nodevalue/client.go:88  [OPCUA] Set to anonymous login
2020-08-11T17:44:36.449+0900    ERROR   nodevalue/nodevalue.go:165      [OPCUA] Tried to connect to OPC UA server 1 time(s). Without success.
2020-08-11T17:44:36.462+0900    INFO    [monitoring]    log/log.go:153  Total non-zero metrics  {"monitoring": {"metrics": {"beat":{"cpu":{"system":{"ticks":156,"time":{"ms":156}},"total":{"ticks":187,"time":{"ms":187},"value":187},"user":{"ticks":31,"time":{"ms":31}}},"handles":{"open":236},"info":{"ephemeral_id":"bd0cdfe6-5384-4376-9e79-4223acf6bc67","uptime":{"ms":2072}},"memstats":{"gc_next":5973680,"memory_alloc":3132680,"memory_total":6589824,"rss":34734080},"runtime":{"goroutines":17}},"libbeat":{"config":{"module":{"running":0}},"output":{"type":"elasticsearch"},"pipeline":{"clients":0,"events":{"active":0}}},"system":{"cpu":{"cores":8}}}}}
2020-08-11T17:44:36.462+0900    INFO    [monitoring]    log/log.go:154  Uptime: 2.0728753s
2020-08-11T17:44:36.463+0900    INFO    [monitoring]    log/log.go:131  Stopping metrics logging.
2020-08-11T17:44:36.464+0900    INFO    instance/beat.go:445    machinebeat stopped.
2020-08-11T17:44:36.465+0900    ERROR   instance/beat.go:933    Exiting: 1 error: Connection was not possible
Exiting: 1 error: Connection was not possible`

felix-lessoer, 2020-08-11:
Are you sure your docker endpoint is supporting anonymous login without any certificate?
Could you also send DEBUG level logs?

dongwoo-john-ku, 2020-08-11:
Hello @felix-lessoer, 
Thank you for your quick reply.

Yes, the endpoint is supporting anonymous login and I could log in the endpoint via uaExpert and a Python library( without any certificate.

I made the endpoint via Prosys OPC UA Simulation Server
[image]

Please refer the below log from exExpert.
`20:26:00.264 | UaExpert           |                                | Loaded Data Access Plugin (Static Plugin)
20:26:00.264 | UaExpert           |                                | Loaded History Plugin (Static Plugin)
20:26:00.264 | UaExpert           |                                | Loaded Events Plugin (Static Plugin)
20:26:00.264 | UaExpert           |                                | Loaded Method Plugin (Static Plugin)
20:26:00.264 | UaExpert           |                                | Loaded Performance Plugin (Static Plugin)
20:26:00.264 | UaExpert           |                                | Loaded Image Viewer Plugin (Static Plugin)
20:26:00.264 | UaExpert           |                                | Loaded Server Diagnostic Plugin (Static Plugin)
20:26:00.264 | UaExpert           |                                | Loaded Data Logger Plugin (Static Plugin)
20:26:00.264 | UaExpert           |                                | Loaded GDS Push Plugin (Static Plugin)
20:26:00.264 | UaExpert           |                                | Loaded File Transfer Plugin (Static Plugin)
20:26:00.264 | UaExpert           |                                | Loaded XML Nodeset Export Plugin (Static Plugin)
20:26:00.275 | UaExpert           |                                | UaExpert is ready to use.
20:26:34.321 | DiscoveryWidget    |                                | Discovery FindServersOnNetwork on opc.tcp://localhost:4840 failed (BadTimeout), falling back to FindServers
20:26:34.824 | DiscoveryWidget    |                                | Discovery FindServers on opc.tcp://localhost:4840 failed (BadTimeout)
20:26:35.326 | DiscoveryWidget    |                                | Discovery GetEndpoints on opc.tcp://localhost:4840 failed (BadTimeout)
20:26:38.857 | Server Node        | opc.tcp://opcuaserver.com:4... | Endpoint: 'opc.tcp://host.docker.internal:53530/OPCUA/SimulationServer'
20:26:38.857 | Server Node        | opc.tcp://opcuaserver.com:4... | Security policy: 'http://opcfoundation.org/UA/SecurityPolicy#None'
20:26:38.857 | Server Node        | opc.tcp://opcuaserver.com:4... | ApplicationUri: 'urn:host.docker.internal:OPCUA:SimulationServer'
20:26:38.857 | Server Node        | opc.tcp://opcuaserver.com:4... | Used UserTokenType: Anonymous
20:26:39.122 | AddressSpaceModel  | opc.tcp://opcuaserver.com:4... | Registered for ModelChangeEvents
20:26:39.122 | Server Node        | opc.tcp://opcuaserver.com:4... | Connection status of server 'opc.tcp://opcuaserver.com:48010' changed to 'Connected'.
20:26:39.122 | Server Node        | opc.tcp://opcuaserver.com:4... | Revised values: SessionTimeout=1200000, SecureChannelLifetime=3600000
20:26:39.126 | AddressSpaceModel  | opc.tcp://opcuaserver.com:4... | Browse on node 'i=84' succeeded.
20:26:39.131 | TypeCache          | opc.tcp://opcuaserver.com:4... | Reading type info of NodeId NS0|Numeric|35 succeeded
20:26:39.137 | TypeCache          | opc.tcp://opcuaserver.com:4... | Reading type info of NodeId NS0|Numeric|33 succeeded
20:26:39.146 | TypeCache          | opc.tcp://opcuaserver.com:4... | Reading type info of NodeId NS0|Numeric|31 succeeded
20:26:39.154 | AddressSpaceModel  | opc.tcp://opcuaserver.com:4... | Browse on node 'i=85' succeeded.
20:26:39.166 | TypeCache          | opc.tcp://opcuaserver.com:4... | Reading type info of NodeId NS0|Numeric|63 succeeded
20:26:39.175 | TypeCache          | opc.tcp://opcuaserver.com:4... | Reading type info of NodeId NS0|Numeric|62 succeeded
20:26:39.181 | TypeCache          | opc.tcp://opcuaserver.com:4... | Reading type info of NodeId NS0|Numeric|2004 succeeded
20:26:39.189 | TypeCache          | opc.tcp://opcuaserver.com:4... | Reading type info of NodeId NS0|Numeric|58 succeeded
`

You may find 'used UserTokenType: Anonymous' and succeeded log result.

I used the Windows latest version of the binary.
 * https://ela.st/machinebeat-windows

Here is the DEBUG level log.

`(base) C:\Users\dwku\OneDrive\바탕 화면\machinebeat>machinebeat.exe -e -c machinebeat.yml
2020-08-11T20:39:26.690+0900    INFO    instance/beat.go:622    Home path: [C:\Users\dwku\OneDrive\바탕 화면\machinebeat] Config path: [C:\Users\dwku\OneDrive\바탕 화면\machinebeat] Data path: [C:\Users\dwku\OneDrive\바탕 화면\machinebeat\data] Logs path: [C:\Users\dwku\OneDrive\바탕  화면\machinebeat\logs]
2020-08-11T20:39:26.752+0900    DEBUG   [beat]  instance/beat.go:674    Beat metadata path: C:\Users\dwku\OneDrive\바탕 화면\machinebeat\data\meta.json
2020-08-11T20:39:26.753+0900    INFO    instance/beat.go:630    Beat ID: d51d9d17-d145-489f-b43d-f9953bc4d771
2020-08-11T20:39:26.818+0900    DEBUG   [filters]       add_cloud_metadata/providers.go:126     add_cloud_metadata: starting to fetch metadata, timeout=3s
2020-08-11T20:39:26.820+0900    DEBUG   [filters]       add_cloud_metadata/providers.go:162     add_cloud_metadata: received disposition for digitalocean after 2.0573ms. result=[provider:digitalocean, error=failed requesting digitalocean metadata: Get http://169.254.169.254/metadata/v1.json: dial tcp 169.254.169.254:80: connectex: A socket operation was attempted to an unreachable network., metadata={}]
[... cut ...] 0,"events":{"active":0}}},"system":{"cpu":{"cores":8}}}}}
2020-08-11T20:39:27.944+0900    INFO    [monitoring]    log/log.go:154  Uptime: 5.3277362s
2020-08-11T20:39:27.945+0900    INFO    [monitoring]    log/log.go:131  Stopping metrics logging.
2020-08-11T20:39:27.947+0900    INFO    instance/beat.go:445    machinebeat stopped.
2020-08-11T20:39:27.947+0900    ERROR   instance/beat.go:933    Exiting: 1 error: Connection was not possible
Exiting: 1 error: Connection was not possible`

Thank you!

felix-lessoer, 2020-08-18:
I was investigating into this issue today.
The problem is, that the prosys server is failing when ApplicationName is not set during the session creation. 
This is the first server that acts like this. 

Unfortunatly the underlying OPC UA library for that beat is also failing when I try to set the ApplicationName. So I need to fix this first to work around the issue in the prosys server.

dongwoo-john-ku, 2020-08-20:
Hello @felix-lessoer,

Thank you for your kind answer.
Please let me know when you release the updated version.

felix-lessoer, 2020-09-02:
@dongwoo-john-ku Prosys released a new version of their server yesterday. The issue that prevents connecting has been fixed now. So connection can be established with that version.

Unfortunatly it looks like this server implementation has more errors. I got invalid data when browsing. I didn't test to directly read the value from specific nodes. That may works.

But in general I would recommend using a better implementation of an OPC UA server.

closed 2020-09-29 by felix-lessoer