elastic/Machinebeat #22: Nodes not being subscribed

elnacho-grande, 2020-08-14:
Hello, 

I started playing around with machinebeat and found this strange behaviour.

Similar to [#20](https://github.com/elastic/Machinebeat/issues/20) I find that it does not matter if enable or disable browsing by `browse.enable: true` or `browse.enable: false` I always see browsing happening in the debug view:

`2020-08-14T14:38:21.487+0200    INFO    nodevalue/nodevalue.go:132      Browsing is enabled. Data collection will start after discovery. Based on your server and browsi
 configuration this can take some time.
2020-08-14T14:38:21.487+0200    INFO    nodevalue/client.go:304 [OPCUA] Start browsing from Objects and Views folder
2020-08-14T14:38:21.487+0200    DEBUG   [Browse]        nodevalue/client.go:341 Start browsing
2020-08-14T14:38:21.487+0200    INFO    nodevalue/client.go:349 Analyse node id i=85
2020-08-14T14:38:21.487+0200    DEBUG   [Get DataType]  nodevalue/client.go:425 This node has no data attached.
2020-08-14T14:38:21.488+0200    DEBUG   [Browse]        nodevalue/client.go:391 Found 6 new nodes for browsing with ref id 0
2020-08-14T14:38:21.488+0200    DEBUG   [Browse]        nodevalue/client.go:341 Start browsing`

Furthermore the nodes that I would like to subscribe are not visible in logstash nor in elasticsearch.
The only events I see are the processor information from machinebeat.yml:

`processors:
  - add_host_metadata: ~
  - add_cloud_metadata: ~`

Here is my opcua.yml file:

`# Module: opcua
# Docs: https://www.elastic.co/guide/en/beats/machinebeat//machinebeat-module-opcua.html

- module: opcua
  metricsets: ["nodevalue"]
  enabled: true
  period: 1s
  
  #The URL of your OPC UA Server
  endpoint: "opc.tcp://localhost:55105"

  #==========================  Security configuration ============================
  #username: ""
  #password: ""
  
  ##If you do not know which policies and modes are supported: start the beat in the debug mode and you will see which is supported by your server
  
  #Possible values: None, Basic128Rsa15, Basic256Sha256, Basic256, Aes128Sha256RsaOaep, Aes256Sha256RsaPss
  #policy: ""
  
  #Possible values: None, Sign, SignAndEncrypt
  #mode: ""

  ##Configuration for certificate based security
  #clientCert: ""
  #clientKey: ""

  ##This needs to be part of the certificates SAN URL and DNS-Name 
  #appName: "machinebeat"
  
  #==========================  Data collection configuration ============================
  
  ##The subscribe mode collects one event whenever there is a change of the value. This is the default.
  ## All collected events will be send after the configured period
  ## If you turn this off the beat will pull the current value after each period.
  #subscribe: true

  ##The browse mode is enabled at default
  ## This means, if the configured nodes have childs all subscribable nodes will be monitored
  ## This does not work if subscribe: false
  browse.enable: false
  #maxLevel: 3
  #maxNodePerParent: 5
  
  ##How often we should retry to connect to the opcua server when something is failes after the first inital successful connection
  retryOnError: 5

  ##How many collect threads should be opened. This is only relevant for non subscribe mode.  
  ## This limit can be reached when it takes longer to get a value than it is configured with period. 
  #maxThreads: 50
  
  #==========================  Node configuration ============================
  ##If this is not configured the browse will start at root. This is required if browse is set to false
  ## Configure the nodes directly to speed up start up of the beat
 nodes:
-  id: "ns=6;s=S71500ET200MP-Station_1.Z105-CPU"
   label: "z105"
`

Might it be because of this following debug message:

`20-08-14T15:41:15.057+0200    DEBUG   [Subscribe]     nodevalue/client.go:255 [OPCUA] Start listening
20-08-14T15:41:15.058+0200    DEBUG   [Subscribe]     nodevalue/client.go:262 [OPCUA] subscription=0 error=The server has reached the maximum number of queued publish requests. StatusBadTooManyPublishRequests (0x80780000)`

Thank you for your help

Yours humbly nacho

felix-lessoer, 2020-08-17:
The right value for the browse configuration is `browse.enabled` .. You didn't set the **d** at the end so it takes the defaults.

elnacho-grande, 2020-08-17:
...well this is embarrasing...it did indeed help with the browsing issue.

Unfortunatley with browsing disabled it publishes not a single event anymore not even the metadata from host.

felix-lessoer, 2020-08-17:
I guess you still see the issue `error=The server has reached the maximum number of queued publish requests. StatusBadTooManyPublishRequests (0x80780000)` ..

So your OPC UA server does not allow more subscriptions. You can increase the limit for that in your OPC UA server.
You may also want to disable subscribe using `subscribe: false` .

elnacho-grande, 2020-08-17:
with disabled browsing ```error=The server has reached the maximum number of queued publish requests. StatusBadTooManyPublishRequests (0x80780000)``` is gone as well

depending on if ```subscribe: false``` I get: 

```2020-08-17T10:38:29.553+0200    DEBUG   [Collector]     nodevalue/nodevalue.go:170      Event collector instance started
2020-08-17T10:38:29.554+0200    DEBUG   [Collect]       nodevalue/client.go:109 Building the request
2020-08-17T10:38:29.554+0200    DEBUG   [Collect]       nodevalue/client.go:144 Sending request
2020-08-17T10:38:29.554+0200    INFO    nodevalue/nodevalue.go:181      error: There was nothing to do because the client passed a list of operations with no elements. StatusBadNothingToDo (0x800F0000) ```

or with ```subscribe: true``` I get: 

```2020-08-17T10:43:15.321+0200    INFO    nodevalue/nodevalue.go:205      [OPCUA] Publishing 0 new events
2020-08-17T10:43:15.321+0200    INFO    nodevalue/nodevalue.go:196      [OPCUA] Too much zero publish attempts.
2020-08-17T10:43:15.321+0200    DEBUG   [Shutdown]      nodevalue/client.go:437 Will shutdown connection savely```

felix-lessoer, 2020-08-17:
Machinebeat is not able to see your nodes configuration. So there is an issue. This configuration file are sensitive for whitespaces.. 

It looks liek the copy here in the issue is missing a <blank> before `-  id: "ns=6;s=S71500ET200MP-Station_1.Z105-CPU"`
The **-** sign needs to be at the same position as the **n** of nodes

The right config would look like:
```
nodes:
-  id: "ns=6;s=S71500ET200MP-Station_1.Z105-CPU"
   label: "z105"
```

elnacho-grande, 2020-08-17:
This did the trick...thank you so much for your help :+1:

closed 2020-08-17 by elnacho-grande