elastic/Machinebeat #25: kepware opcua connection result inquiry.

Buungi, 2020-08-27:
Hello. Thank you for developing machinebeat.

I used machinebeat to connect with kepware opcua.

The connection seems to be fine,
The data to be imported is not the node value that I want.

As shown in the image id: "ns=2;s=Accura/Accura2300/Current"
I want to bring the value
Keep getting the value "opcua.node.ID i=2258".

What should I modify to get the correct value?
Please help me.

opcua.yml

[image]

result

[image]

felix-lessoer, 2020-08-27:
I guess `browse.enabled : false` would help.

If you send the logs I can tell you more.

Buungi, 2020-08-27:
Thanks for the answer.

I've already used that option. "browse.enabled: false"
Even if I try again, it is the same as before.

Below is the log information that I ran.
[image]
[image]

Thank you.

felix-lessoer, 2020-08-27:
As you can see in the logs its still browsing. 
So your browse configuration is not set correctly. 

Make sure that the intention is right.

Buungi, 2020-09-01:
Thank you.

That case was resolved.

browse:
  -enabled: false

This can't get the tag value

browse.enabled: false

This gets the tag value.

closed 2020-09-01 by Buungi