elastic/Machinebeat #28: publishing 0 new events

markusTh83, 2021-03-11:
Hi Felix,

I discovered a problem with the machinebeat, when I set the period quite low (less than 500ms).
When I'm setting the period to 500ms machinebeat works stable and publishes only a few 0 events.
When I'm setting the period to 100ms machinebeat throws (quite regularly) a too much zero publish attempts error and waits for reconnection.  
`2021-03-11T15:53:51.904+0100    INFO    nodevalue/nodevalue.go:215      [OPCUA] Publishing 7 new events
2021-03-11T15:53:52.004+0100    INFO    nodevalue/nodevalue.go:215      [OPCUA] Publishing 0 new events
2021-03-11T15:53:52.103+0100    INFO    nodevalue/nodevalue.go:215      [OPCUA] Publishing 0 new events
2021-03-11T15:53:52.203+0100    INFO    nodevalue/nodevalue.go:215      [OPCUA] Publishing 0 new events
2021-03-11T15:53:52.304+0100    INFO    nodevalue/nodevalue.go:215      [OPCUA] Publishing 0 new events
2021-03-11T15:53:52.404+0100    INFO    nodevalue/nodevalue.go:215      [OPCUA] Publishing 0 new events
2021-03-11T15:53:52.404+0100    INFO    nodevalue/nodevalue.go:206      [OPCUA] Too much zero publish attempts.
2021-03-11T15:53:52.503+0100    INFO    nodevalue/nodevalue.go:315      [OPCUA] Lets wait a while before reconnect happe
ns
2021-03-11T15:53:57.533+0100    INFO    nodevalue/client.go:68  [OPCUA] Get all endpoints from opc.tcp://MyOpcServer:4
840
2021-03-11T15:53:57.839+0100    INFO    nodevalue/client.go:77  [OPCUA] Your selected policy: http://opcfoundation.org/U
A/SecurityPolicy#None and security mode: MessageSecurityModeNone
2021-03-11T15:53:57.839+0100    INFO    nodevalue/client.go:87  [OPCUA] Policy URI: http://opcfoundation.org/UA/Security
Policy#None with security mode MessageSecurityModeNone
2021-03-11T15:53:57.839+0100    INFO    nodevalue/client.go:95  [OPCUA] Set ApplicationName to machinebeat
2021-03-11T15:53:57.839+0100    INFO    nodevalue/client.go:97  [OPCUA] Set ApplicationDescription (SAN DNS and SAN URL)
 to machinebeat
2021-03-11T15:53:57.839+0100    INFO    nodevalue/client.go:110 [OPCUA] Set to anonymous login
2021-03-11T15:53:58.228+0100    INFO    nodevalue/client.go:121 [OPCUA] Connection established
2021-03-11T15:53:58.228+0100    INFO    nodevalue/client.go:204 [OPCUA] Starting subscribe process
2021-03-11T15:53:58.228+0100    INFO    nodevalue/client.go:211 [OPCUA] Subscribe process done
2021-03-11T15:53:58.229+0100    INFO    nodevalue/nodevalue.go:215      [OPCUA] Publishing 0 new events`

felix-lessoer, 2021-03-11:
You are working in subscribe mode. That means that the beat is collecting ALL events WITHIN the period. If you reducing the period this will only effect how often the events get send to Elasticsearch.

If you are looking to get one value per node everytime the period ends.. then you need to deactivate subscribe mode.

markusTh83, 2021-03-11:
Thank you, overseen that....
now it works as a charm!

closed 2021-03-12 by markusTh83