elastic/Machinebeat #29: send data to elastic index

markusTh83, 2021-03-11:
Hi Felix,

one more problem on my side:
I'm not able to write the collected data to an index differing from machinebeat-*

I've tried:
```
setup.template.pattern: "my_index-*"
setup.template.name: "my_index"
```
with a created index mapping:

```
{
  "index_patterns": [
    "my_index-*"
  ],
  "mappings": {
    "_meta": {
      "beat": "my_index",
      "version": "7.10.1"
    },
    "_source": {
      "enabled": false
    },
    "date_detection": false
  },
  "order": 1,
  "settings": {
    "index": {
      "lifecycle": {
        "name": "my_index_rollover",
        "rollover_alias": "my_index_rollover"
      },
      "number_of_shards": 1
    },
      "number_of_shards": 1
  }
}
```
and also with the standard setting of 
```
setup.template.pattern: "machinebeat-*"
setup.template.name: "machinebeat"
```

my output setting is
```
output.elasticsearch:
  # Array of hosts to connect to.
  hosts: ["my.elastic.server:9200"]
  index: "my_index-%{[agent.version]}-%{+yyyy_MM}"

  username: "uname"
  password: "pwd"
```
What am I doing wrong?

Thank you :D

felix-lessoer, 2021-03-26:
Hi @markusTh83 .. Sorry for the late reply. Lost to track it over the weekend and then it got lost.
You need to change the ILM settings like described here: https://www.elastic.co/guide/en/beats/metricbeat/current/ilm.html
If you don't like you need to disable ILM. The index configuration will only be taken if ILM is disabled.

felix-lessoer, 2021-04-12:
Let me know if you need more help, otherwise close the issue, pls.

markusTh83, 2021-04-12:
Hi @felix-lessoer, I've disabled ILM, writing the data to a custom index, and using the elastic ILM on the custom index.
Thank you, it worked :)

closed 2021-04-12 by markusTh83