elastic/Machinebeat #30: Noob question: What to do after MQTT reaches Elasticsearch?

Amalokch, 2021-04-09:
Hello there, first, many thanks for this beat it had helped me to retrieve data from my sensors :-). Problem is that I am pretty new in Elasticsearch and really don't know too much about it, I come from an industrial programming, PLC, SCADA programming, etc.
If I go to analytics -> discovery I can actually see the data. My problem is that it arrives in string format, therefore in Kibana I cannot graph my data. My guess is that I've to convert the data from string to number at some point in the machinebeat.yml file, but honestly I don't know how to do it. do you think you can help me?

Thanks in advance,

felix-lessoer, 2021-04-12:
Hello @Amalokch 
great to hear that you like the beat.

Do you collect the data with MQTT or with OPC UA? Could you also share an example of what you see in Discover? It looks very different per sensor. So its easier to help you knowing the format.

Amalokch, 2021-04-12:
Hello Felix,
Many thanks for your replied. I am collecting the data using the MQTT featur.
I eventually sorted it out. I ended up using a 'processors' to convert the data from string to float, see below a snapshot of my .yml file:

[image]

Now a snapshot of the Discovery with and without this 'processors conversion'

1) without processors
[image]

2) with processors included
[image]

Now I hope you can help me with the following questions :-) :

1) Is it possible to achieve the same without using 'processors'?

2) How can I see the "topic" name in the fields entry? I went through the fields and I cannot see one that says the name of the topic, this becomes more difficult when I use the '# wildcard', since in Kibana I need to split the data into different graphs.
I can see there is a 'metricset.name' which seems to be I can use? but can't figured out how.

3) Lastly, I am pretty new to Ubuntu, and I am using a serverless system. Noticed that when I disconnect via ssh the './machinebeat -e -c machinebeat.yml' command stop working, is there a way to have it running at boot without the need of me typing the above command to run?

So far, thanks in advance for the help.

/Andrés

felix-lessoer, 2021-04-13:
Before I answer your questions I would like to guide you to the official MQTT module of Elastic. Machinebeat was the base for this development: If you would like to get support for your MQTT data collection from Elastic you should use this one: https://www.elastic.co/guide/en/beats/filebeat/current/filebeat-input-mqtt.html

Now let me answer your questions:
1.) Yes there are multiple ways to do so. The Machinebeat MQTT module is trying to decode your payload. This happens when it detects a json structure. If it detects a json structure it also takes care that the fields have the right format.
If you can not switch to JSON you could do it with the processor in the Beat, but usually we use Ingest Node Pipelines in ES for that: https://www.elastic.co/guide/en/elasticsearch/reference/master/ingest.html 
Main advantage is that Kibana offers a GUI to handle that and that your don't need to configure it separatly per beat.

2.) There should be a field called mqtt.topic. To split a Graph you can e.g. use Lens and just drag the field into the diagram area.

3.) I added a machinebeat.service file to systemd to run it as a service
This could look like this:
```
[Info]
Description=Machinebeat to collect MQTT and OPC UA messages

[Service]
User=root
Type=simple
TimeoutSec=0
PIDFile=/var/run/machinebeat.pid
ExecStart=/bin/sh -c '/usr/share/machinebeat/machinebeat -c /usr/share/machinebeat/machinebeat.yml -path.logs /var/log/machinebeat'
KillMode=mixed

Restart=on-failure
RestartSec=42s

StandardOutput=/var/log/machinebeat.log
StandardError=/var/log/machinebeaterr.log

[Install]
WantedBy=default.target

```

Amalokch, 2021-04-13:
Thanks Felix!

closed 2021-04-13 by Amalokch