elastic/Machinebeat #31: Windows version of machinebeat has a malicious file

Valdos116, 2021-06-23:
The windows version of the machinebeat has a malicious file flagged by Jiangmin scanner software. Please note that two other sites came up with the same result as well.

A new compiled version is required in order to use machinebeat.

felix-lessoer, 2021-06-24:
I've updated the dependencies and released a new version based on the latest beats framework. 
Now virustotal is not showing a malicious file.

Valdos116, 2021-06-24:
@Felix-lessoer, can you point me to the link to download the new version?

felix-lessoer, 2021-06-24:
Sure. You can use the "release" section in github: https://github.com/elastic/Machinebeat/releases/tag/v7.13.2
Or you can download it from the fileserver: https://ela.st/machinebeat-windows

Valdos116, 2021-06-24:
Thank you so much. I'm pretty new to Elastic.I wanted to use machinebeat to collect data from industrial machinery with data communications protocol like : opc,mqtt,modbus,bacnet etc..I'm using puTTy interface to set up these drivers to collect data from theses PLC-based devices.
-Do you have a PuTTy set of command list(documentation) to accomplish this?
-Do you think a filebeat should be enough to interface with Modbus and BACnet drivers?

felix-lessoer, 2021-06-24:
This beat so far only collects via MQTT and OPC UA. These two reflecting the most common use cases. Which one to use depends on the machine you want to get data from.
To get data via modbus or bacnet you need an interface, e.g. https://www.cybus.io/en/ . They can collect data using many different protocols and translate it into MQTT messages which we can collect.

The beat usually listening to a specific endpoint url. All you need to now you can find in the readme and config file samples that are part of the release. If you have a specific connection problem open a new ticket, pls.

Is it okay to close this issue?

Valdos116, 2021-06-24:
Thanks.
Yes you can close this issue!

closed 2021-06-24 by Valdos116