elastic/Machinebeat #32: Ingest data via Modbus and BACnet communication protocols to Elastic

Valdos116, 2021-07-02:
Hello Team,

Trying to ingest data from Modbus TCP and BACnet IP devices into Logstash. Seems like there is no driver so far to accomplish this since machinebeat only communicates over OPC UA.

Any help would be appreciated

Thanks

felix-lessoer, 2021-07-02:
You need to use some kind of interface software to get this done..
A good one is https://www.cybus.io/en/

Googeling for modbus to mqtt or opcua will you show more results

Valdos116, 2021-07-02:
Thanks! Is this cybus interface software easily integrable with Elastic?

felix-lessoer, 2021-07-02:
Yes, they have an easy to use elastic integration translating all their industry protocols into MQTT. The MQTT data gets collected from Filebeat MQTT input.
We just made a webinar together with them to show the value. This are all the protocols they support https://docs.cybus.io/docs/1.0.46/user/protocols/index.html

closed 2021-07-13 by felix-lessoer

Valdos116, 2021-07-13:
Hello Guys,
Machinebeat communication lost due to power outage. Trying to restart it but for some reasons it won't make a connection and it's reporting nothing in the log file. Do you know how to get the response code from machinebeat when the connection failed?

felix-lessoer, 2021-07-13:
Can you create a new issue and add logs pls? You could also enable debug logs. That will add more information of connection is failing.

Valdos116, 2021-07-13:
Never mind . The issue is solved.Thanks