elastic/Machinebeat #35: Apache Kafka Output possible?

totoroot, 2021-07-15:
The README mentions that Apache Kafka can be used to process the message queue instead of connecting directly to an Elasticsearch or Logstash instance. Now how would one go about configuring a Kafka output? The machinebeat.yml configuration file has entries for the Elasticsearch and Logstash outputs but none for Kafka.

I've successfully configured the OPC input module but how would one go about configuring the Kafka output module?

Can it be configured just like for metricbeat?
https://www.elastic.co/guide/en/beats/metricbeat/current/metricbeat-module-kafka.html

Also I noticed that the [official link to the documentation](https://www.elastic.co/guide/en/beats/machinebeat//machinebeat-module-opcua.html) in the [OPC-UA module configuration file](https://github.com/elastic/Machinebeat/blob/master/modules.d/opcua.yml.disabled) is no longer valid.

felix-lessoer, 2021-07-15:
Thanks for testing Machinebeat. We didn't release any official documentation so far. The link is auto generated and never led to any documentation.
The Machinebeat is more or less a copy of Metricbeat just with the Industry related protocols to get data from.
So the documentation from Metricbeat for everything that does not belong to data collection modules should be the same.

Saying that.. also the Kafka Output most likely works, but I never tested it before. Would be nice if you could just test it and provide your results here.

felix-lessoer, 2021-08-13:
@totoroot is it okay to close the issue?

totoroot, 2021-08-23:
@felix-lessoer I haven't yet been able to test it thoroughly, but it seems to work. If any other problems turn up, I will open another issue. Thanks for your response!!

closed 2021-08-23 by totoroot