elastic/Machinebeat #36: security configuration with Aspentech IP21

rohyoungsu, 2021-07-20:
I am trying to connect to Aspentech IP21 OPC UA server.
They sent certificate file as "xxxx.der". but I don't know how to use it in machinebeat.
Can i get some guide about it?

rohyoungsu, 2021-07-20:
IP21 guy said it is a kind of certificate file which is used in windows system to verify trust computer.
and I set this certificate as below but, it didn't work well.

- module: opcua
  metricsets: ["nodevalue"]
  enabled: true
  period: 1s

  endpoint: "opc.tcp://192.168.10.134:63500/InfoPlus21/OpcUa/Server"

  username: "skynet.help\rtdbapp"
  password: "passw0rd"

  securityMode: "SignAndEncrypt"
  clientCert: "/home/elastic/AspenTech-InfoPlus.21-OPC-UA-SERVER-ABCD3455DFG332525235243232432423.der"

felix-lessoer, 2021-07-20:
Be careful sharing your endpoint + credentials, pls. This repo is public.

I think you need to transform the file into the expected format. I can give you a hint later today.

rohyoungsu, 2021-07-20:
unfortunately, I can't attach file because of security system in my company.
I have converted .der file into .pem with openssl and tried again. but it was failed.
if you have any idea to trasform certificate file please let me know.

felix-lessoer, 2021-07-20:
For PEM files you need to set `clientCert `and `clientKey `.. Did you configured both files?
Also its important to set the `appName`. Do make your certs valid the `appName `configuration needs to be available within your certificate as SAN URL and DNS-Name

You can have a look into your .der file by double clicking on it:
[image]

rohyoungsu, 2021-07-21:
I did set one pem file to both clientCert and clientKey and I find out DNS-Nama as "C03VP-RTDB-DB1" and I did set appName as "C03VP-RTDB-DB1" and try again and machinebeat showed me as "opcua: Failed to decode PEM block with private key"

rohyoungsu, 2021-07-21:
[image]
It is my pem and when I convert der to pem it generate just one pem file

rohyoungsu, 2021-07-21:
I  request a private key file to AspenTech guy. I will let you know test result after get the private key file from AspenTech.
Thank you very much

rohyoungsu, 2021-07-22:
AspenTech reject my request. So I will find out other way to collect data from AspenTech IP21.
Anyway, I have to say I am sorry because first time I didn't expect your response on time.
And thank you very much.

closed 2021-07-22 by rohyoungsu