elastic/Machinebeat #37: Future support/logstash alternative

robbertvanhove, 2021-08-04:
Hello

As said in the first lines of the README, Machinebeat is experimental & not recommended to use in production. But my client would like to get the data from their OPC UA server into ElasticSearch. 

As an alternative, I was thinking about using Logstash to read directly from the OPC UA server. Is this possible via some kind of logstash plugin? How is this done in other applications that use OPC UA? 

Thx!

felix-lessoer, 2021-08-04:
Hello

there is no plugin for Logstash that can handle OPC UA.
However there are many users that collect their OPC UA data using Machinebeat even in production. The disclaimer is there to make you aware that this is not supported by Elastic Support contracts and if something breaks that only myself can help you.. But it is the easiest way to get OPC UA data into Elasticsearch.
Its your decision to use it in any way you like.

felix-lessoer, 2021-08-13:
@robbertvanhove is it okay to close the issue?

robbertvanhove, 2021-08-13:
@felix-lessoer yes! Thx your answer was straight to the point.

closed 2021-08-13 by felix-lessoer