elastic/Machinebeat #38: OPC-UA: Crash after certain amount of time 'panic: close of closed channel'

robbertvanhove, 2021-08-18:
After a certain amount of time without a value change on the subscribed tag, the following happens:

Machinebeat output: 
```
panic: close of closed channel

goroutine 656 [running]:
github.com/gopcua/opcua/uasc.(*SecureChannel).Close.func1(0xc0003de700)
        C:/Users/felix/Documents/Machinebeat/pkg/mod/github.com/gopcua/opcua@v0.2.0-rc1/uasc/secure_channel.go:842 +0x36
github.com/gopcua/opcua/uasc.(*SecureChannel).Close(0xc0003de700, 0x2e8a2a0, 0xc000094110)
        C:/Users/felix/Documents/Machinebeat/pkg/mod/github.com/gopcua/opcua@v0.2.0-rc1/uasc/secure_channel.go:851 +0x1a5
github.com/gopcua/opcua.(*Client).monitor(0xc00030c7e0, 0x2ecced8, 0xc0001289c0)
        C:/Users/felix/Documents/Machinebeat/pkg/mod/github.com/gopcua/opcua@v0.2.0-rc1/client.go:326 +0x1006
created by github.com/gopcua/opcua.(*Client).Connect.func1
        C:/Users/felix/Documents/Machinebeat/pkg/mod/github.com/gopcua/opcua@v0.2.0-rc1/client.go:216 +0x6a
```

opcua.yml:
```
- module: opcua
  metricsets: ["nodevalue"]
  enabled: true
#   period: 1s
  
  #The URL of your OPC UA Server
  endpoint: "MYURL"

  #==========================  Security configuration ============================
  username: "MYUSER"
  password: "MYPASS"
  

  
  #==========================  Data collection configuration ============================

  ecsFields: true
  legacyFields: false  

  subscribe: true

  browse.enabled: false

  
  nodes:
  -  id: "ns=2;s=Path.To.TestBoolean"
     label: "TestBoolean"
```

Am I doing something wrong?

magiconair, 2022-01-27:
We've fixed a number of issues especially in the 0.2.x codebase so I suggest to upgrade to v0.3.1 which I'm about to release. v0.2.0-rc1 has a lot of issues.

felix-lessoer, 2022-01-27:
Thanks @magiconair . I will do that for the next version of the beat.

felix-lessoer, 2022-02-09:
Upgraded to latest version of gopcua. This will most likely fix this issue. @robbertvanhove Can you confirm?

closed 2022-02-22 by felix-lessoer