elastic/Machinebeat #42: Panic: runtime error: index out of range [33] with length 20

ismyhairnice, 2022-02-09:
Hi all

Machinebeat encounter panic runtime error when i let it run for several hours. Below is the error message:

```
2022-02-09T03:05:34.038+0800	INFO	nodevalue/nodevalue.go:217	[OPCUA] Publishing 0 new events
2022-02-09T03:05:36.038+0800	INFO	nodevalue/nodevalue.go:217	[OPCUA] Publishing 0 new events
2022-02-09T03:05:38.038+0800	INFO	nodevalue/nodevalue.go:217	[OPCUA] Publishing 0 new events
2022-02-09T03:05:40.038+0800	INFO	nodevalue/nodevalue.go:217	[OPCUA] Publishing 0 new events
2022-02-09T03:05:42.038+0800	INFO	nodevalue/nodevalue.go:217	[OPCUA] Publishing 0 new events
2022-02-09T03:05:44.038+0800	INFO	nodevalue/nodevalue.go:217	[OPCUA] Publishing 0 new events
2022-02-09T03:05:46.038+0800	INFO	nodevalue/nodevalue.go:217	[OPCUA] Publishing 0 new events
panic: runtime error: index out of range [33] with length 20

goroutine 1139 [running]:
github.com/elastic/machinebeat/module/opcua/nodevalue.(*Client).subscribeTo(0xc0002ee880)
	C:/Users/FelixRoessel/Documents/Projekte/Machinebeat/src/github.com/elastic/machinebeat/module/opcua/nodevalue/client.go:286 +0xc32
created by github.com/elastic/machinebeat/module/opcua/nodevalue.(*Client).startSubscription
	C:/Users/FelixRoessel/Documents/Projekte/Machinebeat/src/github.com/elastic/machinebeat/module/opcua/nodevalue/client.go:209 +0x74
```
May I know what is causing this error?

felix-lessoer, 2022-02-09:
This can happen if the subscription gets canceled from the server without notifying the client.
I've upgraded the client now. This includes improved handling of subscriptions. 

Can you confirm that the latest version is fixing this issue for you?

ismyhairnice, 2022-02-10:
Hi Felix

I let the machinebeat ran for past 12 hours and seems that there is no crashes.

Thanks!

ismyhairnice, 2022-02-14:
Hi there

It seems that after a while, the panic happens again

2022-02-10T23:43:01.509+0800	INFO	nodevalue/nodevalue.go:208	[OPCUA] Too much zero publish attempts.
panic: close of closed channel

Is there anything need to be done on my side?

felix-lessoer, 2022-02-14:
Can you share a bit more of the logs?
the zero publish event trigger occurs when there was no data for quite some time.
This can happen if the subscription got closed from server side.
If that happens the beat should reconnect. Thats what I would expect to happen after the handled panic above.

ismyhairnice, 2022-02-15:
I found out that the panic: close of closed channel error is due to the server side application went offline. The beat didn't seems to restart after that:

panic: close of closed channel

goroutine 27123 [running]:
github.com/gopcua/opcua/uasc.(*SecureChannel).Close.func1(0xc0002881c0)
	C:/Users/FelixRoessel/Documents/Projekte/Machinebeat/pkg/mod/github.com/gopcua/opcua@v0.2.0-rc1/uasc/secure_channel.go:842 +0x2f
github.com/gopcua/opcua/uasc.(*SecureChannel).Close(0xc0002881c0, 0x2419c60, 0xc000102030)
	C:/Users/FelixRoessel/Documents/Projekte/Machinebeat/pkg/mod/github.com/gopcua/opcua@v0.2.0-rc1/uasc/secure_channel.go:851 +0x1a5
github.com/gopcua/opcua.(*Client).monitor(0xc0005bc000, 0x2465460, 0xc00065cf40)
	C:/Users/FelixRoessel/Documents/Projekte/Machinebeat/pkg/mod/github.com/gopcua/opcua@v0.2.0-rc1/client.go:326 +0xfe6
created by github.com/gopcua/opcua.(*Client).Connect.func1
	C:/Users/FelixRoessel/Documents/Projekte/Machinebeat/pkg/mod/github.com/gopcua/opcua@v0.2.0-rc1/client.go:216 +0x65

felix-lessoer, 2022-02-16:
For an complete check whats happening you sended not enough of the logs. This is just info about the framework not about the code that is triggering the framework.
However I can see that the version you are using is still using the old version of the framework. The newer version should be able to handle those failure states much better.
I've just uploaded a new build of the same version for the beat.
Can you try re-downloading the beat and run again?
https://ela.st/machinebeat-windows
https://ela.st/machinebeat-linux

closed 2022-02-22 by felix-lessoer