elastic/Machinebeat #43: [Question] Support "Word" data type

softflowinc, 2022-02-21:
Dear Contributor,

Thanks for your work.
I could integrate my OPC UA Server into ELK stack.
But, I figured out that only Bool data type can be added automatically.

> 2022-02-21T17:44:58.352+0900    INFO    nodevalue/nodevalue.go:130      Browsing is enabled. Data collection will start after discovery. Based on your server and browsing configuration this can take some time.
> 2022-02-21T17:44:58.352+0900    INFO    nodevalue/client.go:309 [OPCUA] Start browsing node: ns=4;i=1
> 2022-02-21T17:44:58.352+0900    INFO    nodevalue/client.go:359 Analyse node id ns=4;i=1
> 2022-02-21T17:44:58.376+0900    INFO    nodevalue/client.go:359 Analyse node id ns=4;i=25
> 2022-02-21T17:44:58.385+0900    INFO    nodevalue/client.go:381 Add new node to list: ID: ns=4;i=25| Type bool| Name variable1
> 2022-02-21T17:44:58.394+0900    INFO    nodevalue/client.go:359 Analyse node id ns=4;i=26
> 2022-02-21T17:44:58.403+0900    INFO    nodevalue/client.go:381 Add new node to list: ID: ns=4;i=26| Type bool| Name variable2
> 2022-02-21T17:44:58.412+0900    INFO    nodevalue/client.go:359 Analyse node id ns=4;i=27
> 2022-02-21T17:44:58.431+0900    INFO    nodevalue/client.go:359 Analyse node id ns=4;i=28
> 2022-02-21T17:44:58.449+0900    INFO    nodevalue/client.go:359 Analyse node id ns=4;i=29

ns=4;i=28 NodeID is WORD type variabls.
Could you anyone help me to add node regarding of "WORD" type?
Here's OPC UA configuration applied.

> # Module: opcua
> # Docs: https://www.elastic.co/guide/en/beats/machinebeat//machinebeat-module-opcua.html
> 
> - module: opcua
>   metricsets: ["nodevalue"]
>   enabled: true
>   period: 1s
> 
>   #The URL of your OPC UA Server
>   endpoint: "opc.tcp://192.168.1.202:4840"
> 
>   #==========================  Security configuration ============================
>   #username: ""
>   #password: ""
> 
>   ##If you do not know which policies and modes are supported: start the beat in the debug mode and you will see which is supported by your server
> 
>   #Possible values: None, Basic128Rsa15, Basic256Sha256, Basic256, Aes128Sha256RsaOaep, Aes256Sha256RsaPss
>   #policy: ""
> 
>   #Possible values: None, Sign, SignAndEncrypt
>   #mode: ""
> 
>   ##Configuration for certificate based security
>   #clientCert: ""
>   #clientKey: ""
> 
>   ##This needs to be part of the certificates SAN URL and DNS-Name
>   #appName: "machinebeat"
> 
>   #==========================  Data collection configuration ============================
> 
>   ##The subscribe mode collects one event whenever there is a change of the value. This is the default.
>   ## All collected events will be send after the configured period
>   ## If you turn this off the beat will pull the current value after each period.
>   #subscribe: true
> 
>   ##The browse mode is enabled at default
>   ## This means, if the configured nodes have childs all subscribable nodes will be monitored
>   browse.enabled: true
>   browse.maxLevel: 100
>   browse.maxNodePerParent: 100
> 
>   ##How often we should retry to connect to the opcua server when something is failes after the first inital successful connection
>   #retryOnError: 5
> 
>   ##How many collect threads should be opened. This is only relevant for non subscribe mode.
>   ## This limit can be reached when it takes longer to get a value than it is configured with period.
>   #maxThreads: 50
> 
>   #==========================  Node configuration ============================
>   ##If this is not configured the browse will start at root. This is required if browse is set to false
>   ## Configure the nodes directly to speed up start up of the beat
>   nodes:
>   -  id: "ns=4;i=1"
>      label: "OPC UA"

Thanks in advance.

felix-lessoer, 2022-02-21:
Hi @softflowinc 

this are the supported DataTypes. So String (word) should be loaded. 
[image]

Can you run in with log level debug and send the logs again? I may be able to see more then.

softflowinc, 2022-02-21:
Hi @felix-lessoer 

Thanks for the reply.

I've added option in "machine.yaml"
```
#================================ Logging =====================================

# Sets log level. The default log level is info.
# Available log levels are: error, warning, info, debug
logging:
  - level: debug
```

and, I've got result like below.
```
2022-02-21T18:47:00.449+0900    INFO    instance/beat.go:686    Home path: [/home/odroid/Documents/workspace/Machinebeat] Config path: [/home/odroid/Documents/workspace/Machinebeat] Data path: [/home/odroid/Documents/workspace/Machinebeat/data] Logs path: [/home/odroid/Documents/workspace/Machinebeat/logs] Hostfs Path: [/]
2022-02-21T18:47:00.449+0900    INFO    instance/beat.go:694    Beat ID: 7a3790de-5277-4b92-8fca-a9d72423d13d
2022-02-21T18:47:03.455+0900    WARN    [add_cloud_metadata]    add_cloud_metadata/provider_aws_ec2.go:79       read token request for getting IMDSv2 token returns empty: Put "http://169.254.169.254/latest/api/token": context deadline exceeded (Client.Timeout exceeded while awaiting headers). No token in the metadata request will be used.
2022-02-21T18:47:03.460+0900    INFO    [seccomp]       seccomp/seccomp.go:124  Syscall filter successfully installed
2022-02-21T18:47:03.460+0900    INFO    [beat]  instance/beat.go:1040   Beat info       {"system_info": {"beat": {"path": {"config": "/home/odroid/Documents/workspace/Machinebeat", "data": "/home/odroid/Documents/workspace/Machinebeat/data", "home": "/home/odroid/Documents/workspace/Machinebeat", "logs": "/home/odroid/Documents/workspace/Machinebeat/logs"}, "type": "machinebeat", "uuid": "7a3790de-5277-4b92-8fca-a9d72423d13d"}}}
2022-02-21T18:47:03.460+0900    INFO    [beat]  instance/beat.go:1049   Build info      {"system_info": {"build": {"commit": "unknown", "libbeat": "7.17.0", "time": "1754-08-30T22:43:41.128Z", "version": "7.17.0"}}}
2022-02-21T18:47:03.461+0900    INFO    [beat]  instance/beat.go:1052   Go runtime info {"system_info": {"go": {"os":"linux","arch":"arm","max_procs":8,"version":"go1.17.6"}}}
2022-02-21T18:47:03.463+0900    INFO    [beat]  instance/beat.go:1056   Host info       {"system_info": {"host": {"architecture":"armv7l","boot_time":"2022-02-21T15:47:11+09:00","containerized":false,"name":"odroid","ip":["127.0.0.1/8","::1/128","192.168.1.188/24","fdb2:26bc:7614::bd8/128","fdb2:26bc:7614:0:3db1:88c5:7ade:6e38/64","fdb2:26bc:7614:0:675a:311c:5785:3c02/64","fe80::380e:b25b:2090:6c0f/64","192.168.214.23/24","fe80::9a83:9132:13f1:7588/64","172.17.0.1/16"],"kernel_version":"5.4.167-240","mac":["00:1e:06:37:1f:a2","20:0d:b0:4b:89:ca","02:42:f8:13:d3:38"],"os":{"type":"linux","family":"debian","platform":"ubuntu","name":"Ubuntu","version":"20.04.4 LTS (Focal Fossa)","major":20,"minor":4,"patch":4,"codename":"focal"},"timezone":"KST","timezone_offset_sec":32400,"id":"785002f4a3aeb23d497f413f618dec80"}}}
2022-02-21T18:47:03.465+0900    INFO    [beat]  instance/beat.go:1085   Process info    {"system_info": {"process": {"capabilities": {"inheritable":null,"permitted":null,"effective":null,"bounding":["chown","dac_override","dac_read_search","fowner","fsetid","kill","setgid","setuid","setpcap","linux_immutable","net_bind_service","net_broadcast","net_admin","net_raw","ipc_lock","ipc_owner","sys_module","sys_rawio","sys_chroot","sys_ptrace","sys_pacct","sys_admin","sys_boot","sys_nice","sys_resource","sys_time","sys_tty_config","mknod","lease","audit_write","audit_control","setfcap","mac_override","mac_admin","syslog","wake_alarm","block_suspend","audit_read"],"ambient":null}, "cwd": "/home/odroid/Documents/workspace/Machinebeat", "exe": "/home/odroid/Documents/workspace/Machinebeat/machinebeat", "name": "machinebeat", "pid": 14779, "ppid": 4073, "seccomp": {"mode":"filter","no_new_privs":true}, "start_time": "2022-02-21T18:46:59.760+0900"}}}
2022-02-21T18:47:03.465+0900    INFO    instance/beat.go:328    Setup Beat: machinebeat; Version: 7.17.0
2022-02-21T18:47:03.466+0900    INFO    [publisher]     pipeline/module.go:113  Beat name: odroid
2022-02-21T18:47:03.468+0900    INFO    instance/beat.go:492    machinebeat start running.
2022-02-21T18:47:03.468+0900    INFO    [monitoring]    log/log.go:142  Starting metrics logging every 30s
2022-02-21T18:47:03.475+0900    WARN    [cfgwarn]       nodevalue/nodevalue.go:90       BETA: The OPCUA metricset is beta.
2022-02-21T18:47:03.476+0900    INFO    nodevalue/client.go:68  [OPCUA] Get all endpoints from opc.tcp://192.168.1.202:4840
2022-02-21T18:47:03.517+0900    INFO    [add_cloud_metadata]    add_cloud_metadata/add_cloud_metadata.go:101    add_cloud_metadata: hosting provider type not detected.
2022-02-21T18:47:03.521+0900    INFO    nodevalue/client.go:77  [OPCUA] Your selected policy: http://opcfoundation.org/UA/SecurityPolicy#None and security mode: MessageSecurityModeNone
2022-02-21T18:47:03.521+0900    INFO    nodevalue/client.go:87  [OPCUA] Policy URI: http://opcfoundation.org/UA/SecurityPolicy#None with security mode MessageSecurityModeNone
2022-02-21T18:47:03.522+0900    INFO    nodevalue/client.go:95  [OPCUA] Set ApplicationName to machinebeat
2022-02-21T18:47:03.522+0900    INFO    nodevalue/client.go:97  [OPCUA] Set ApplicationDescription (SAN DNS and SAN URL) to machinebeat
2022-02-21T18:47:03.522+0900    INFO    nodevalue/client.go:110 [OPCUA] Set to anonymous login
2022-02-21T18:47:03.583+0900    INFO    nodevalue/client.go:121 [OPCUA] Connection established
[... cut ...]
2022-02-21T18:47:04.091+0900    INFO    nodevalue/client.go:242 [OPCUA] Add node to subscription: ns=4;i=26
2022-02-21T18:47:04.101+0900    INFO    nodevalue/client.go:242 [OPCUA] Add node to subscription: ns=4;i=30
2022-02-21T18:47:05.064+0900    INFO    nodevalue/nodevalue.go:214      [OPCUA] Publishing 0 new events
2022-02-21T18:47:06.064+0900    INFO    nodevalue/nodevalue.go:214      [OPCUA] Publishing 3 new events
2022-02-21T18:47:07.065+0900    INFO    nodevalue/nodevalue.go:214      [OPCUA] Publishing 0 new events
```

What I am trying to add is "WORD" data type literally.
[image]

felix-lessoer, 2022-02-21:
It didn't took the debug level for the logs. Its still on info. The configuration seems to be fine..
But the right file would be `machinebeat.yml` instead of `machine.yaml` .

Can you check?

mateoricfarm, 2022-02-21:
[image]

I debugged it and figured out that variable "v" is "0xbba" which is not in the switch~case statement.
I don't know what "0xbba" means, but "getDataType" func looks like refrenced code by "gopuca"
I'll look through the "gopcua" project.

Sincerely,

closed 2022-02-21 by mateoricfarm

felix-lessoer, 2022-02-21:
Did you found the issue?

Otherwise you could add **vvv** to the default of switch case
`def.DataType = attrs[4].Value.NodeID().String()`

mateoricfarm, 2022-02-21:
@felix-lessoer,

Yes, I'll give it a try exactly as you mentioned.

felix-lessoer, 2022-02-21:
Would be great to know if it works like this. 
If yes I will add the change to the main version

mateoricfarm, 2022-02-21:
[image]

I was able to add node where there is non-default datatype by adding code below like you mentioned.
```
default:
    return value.Value.NodeID().String()
```

Thank you again.