elastic/Machinebeat #44: Machinebeat logs having statement: "is something else entirely"

pankaj-k, 2022-02-22:
Hi,
I am reading this data off Kepware. 
When I run the machinebeat (I have not yet configured a elasticsearch details into the machinebeat.yml) to see if it is able to read data in the first place. I see this:

2022-02-22T11:08:53.965+1100    INFO    nodevalue/nodevalue.go:217      [OPCUA] Publishing 15 new events
21.30145263671875 is something else entirely
25 is something else entirely
21.40155029296875 is something else entirely
21.601648330688477 is something else entirely
21.40155029296875 is something else entirely
21.501453399658203 is something else entirely
21.101451873779297 is something else entirely
21.40155029296875 is something else entirely
21.40155029296875 is something else entirely
21.301549911499023 is something else entirely
17.165603637695312 is something else entirely

This behaviour is seen in the latest verstion 7.17.0

In version 7.13.2 this is not seen. It shows something like this:
2022-02-22T11:46:22.113+1100    INFO    nodevalue/nodevalue.go:214      [OPCUA] Publishing 1 new events
2022-02-22T11:46:23.112+1100    INFO    nodevalue/nodevalue.go:214      [OPCUA] Publishing 2 new events
2022-02-22T11:46:24.118+1100    INFO    nodevalue/nodevalue.go:214      [OPCUA] Publishing 1 new events
2022-02-22T11:46:25.111+1100    INFO    nodevalue/nodevalue.go:214      [OPCUA] Publishing 2 new events
2022-02-22T11:46:26.115+1100    INFO    nodevalue/nodevalue.go:214      [OPCUA] Publishing 0 new events

Changes in opcua.yml are:
  #The URL of your OPC UA Server
  endpoint: "opc.tcp://ourserver:49320"

  nodes:
  -  id: "ns=2;s=demo.fixture.readings"
     label: "fixture"

felix-lessoer, 2022-02-22:
Hi @pankaj-k 

this is an additional check for some very specific data types. Looks like I didn't made this a debug message.
You can safely ignore it. Will be removed for the next version.

Thanks for using Machinebeat

pankaj-k, 2022-02-22:
Hi Felix,

Thanks for the prompt response. Is it possible to get an approximate time for the release of the next version?

Regards
Pankaj K

felix-lessoer, 2022-02-22:
Can do it anytime. Only takes a few minutes

pankaj-k, 2022-02-22:
Hi Felix,

Can I request you to do a release as I am planning to use it one of the systems we are developing. 

Regards
Pankaj K

felix-lessoer, 2022-02-23:
Done

closed 2022-02-23 by felix-lessoer