elastic/Machinebeat #45: sensor.label and sensor.name having same value in Kibana

pankaj-k, 2022-02-23:
Hi,

I am not sure if this is expected but since there is an option of providing the label value in the opcua.yml I thought it label should be configurable by the user. Or it might be that I am looking at wrong field in Kibana. 

Here is the opcua.yml:

```
- module: opcua
  metricsets: ["nodevalue"]
  enabled: true
  period: 1s
  
  #The URL of your OPC UA Server
  endpoint: "opc.tcp://thecompanyserver:49320"

  #browse.enabled: true

  nodes:
  -  id: "ns=2;s=Asset1.Block.Oven"
     label: "Oven"
  -  id: "ns=2;s=Asset1.Block.Robot"
     label: "Robot"
```
There are many tags (in Kepware speak) or variables or nodes (I am a newbie not familiar with OPC UA language) under both Asset1.Block.Oven as well as Asset1.Block.Robot node.

```
Project
  -- Connections
        -- Asset1
              -- Block
                     -- Oven
                         -- TC1
                         -- TC2
                         -- TC3
                     -- Robot
                         -- Angle1
                         -- Angle2
                         -- Angle3

```
Suppose there is a variable or tag (in Kepware speak) under Asset.One.Block.Oven called TC1. 

My expectation was that I will see in Kibana:
sensor.name: TC1
sensor.label: **Oven**
value.value_float64: 123.23

However in Kibana I see:
sensor.name: TC1
sensor.label: **TC1**
value.value_float64: 123.23

Is it because there are nested elements/tags/variables under the nodes I have specified? I was hoping that they will be auto discovered and will have labels on them. 

Machinebeat Version: v7.17.1, v7.17.0
Kibana: 7.16.3
Logstash: 7.13.2

Regards
Pankaj K

felix-lessoer, 2022-02-23:
The labels will be used if you disable browsing.
Whenever you browse the beat takes the node name as label for the node. There is no hand over from higher hierarchy levels into the lower ones.

So what you can do is: 
1.) Adding the nodes (lowest child) and defining the labels manually in the node list.
2.) Use another field that knows the parent
2b) Use that field to override the value in the way you expect it using Ingest Node Pipelines or runtime fields.

pankaj-k, 2022-02-23:
Ah I got it. 
No worries. I am getting 
sensor.id : ns=2;s=Asset1.Block.Oven.TC1 
which I can split to get the value "Oven" to overwrite into the sensor.label field.

Maybe it can be a nifty feature to be implemented in the next release provided it does not fall foul of the OPC UA protocol. However I am already seeing edge cases where the nesting is more than one level deep. 

Thanks again for prompt response.

Also the current release has version 7.17.0 instead of 7.17.1 in sensor.version.

closed 2022-08-15 by felix-lessoer