elastic/Machinebeat #46: Failed to connect to backoff: Get EOF

BigeYoung, 2022-04-08:
machinebeat is running on **my computer**
- IP: 192.168.137.120
- OS: Windows11 21H2

I am sure that I can connect to the server from **my computer**: I use [Elasticvue](https://microsoftedge.microsoft.com/addons/detail/geifniocjfnfilcbeloeidajlfmhdlgo) and it works well.

I also checked by `curl`:

```ps
curl http://192.168.137.121:9200/

StatusCode        : 200
StatusDescription : OK
Content           : {
                      "name" : "server1",
                      "cluster_name" : "cpps-server",
                      "cluster_uuid" : "Cdy1jqpYTyWRWsHjU6NebQ",
                      "version" : {
                        "number" : "7.17.2",
                        "build_flavor" : "default",
                        "build_type" : "rpm"...
RawContent        : HTTP/1.1 200 OK
                    X-elastic-product: Elasticsearch
                    Warning: 299 Elasticsearch-7.17.2-de7261de50d90919ae53b0eff9413fd7e5307301 "Elast
                    icsearch built-in security features are not enabled. Without authent...
Forms             : {}
Headers           : {[X-elastic-product, Elasticsearch], [Warning, 299 Elasticsearch-7.17.2-de7261de5
                    0d90919ae53b0eff9413fd7e5307301 "Elasticsearch built-in security features are not
                     enabled. Without authentication, your cluster could be accessible to anyone. See
                     https://www.elastic.co/guide/en/elasticsearch/reference/7.17/security-minimal-se
                    tup.html to enable security."], [Content-Length, 535], [Content-Type, application
                    /json; charset=UTF-8]}
Images            : {}
InputFields       : {}
Links             : {}
ParsedHtml        : mshtml.HTMLDocumentClass
RawContentLength  : 535
```

I have turned off all firewalls on `server1` and `my computer`.

machinebeat configuration `machinebeat.yml`:

```yml
#==========================  Modules configuration ============================

machinebeat.config.modules:
  path: ${path.config}/modules.d/*.yml
  reload.enabled: false

#==================== Elasticsearch template setting ==========================

setup.template.settings:
  index.number_of_shards: 1
  index.codec: best_compression
  
#================================ Outputs =====================================
#-------------------------- Elasticsearch output ------------------------------
output.elasticsearch:
  hosts: ["192.168.137.121"]

#================================ Processors =====================================
processors:
  - add_host_metadata: ~
  - add_cloud_metadata: ~

#================================ Logging =====================================
logging.level: debug
```

------

test:

```bash
.\machinebeat.exe test output -c .\machinebeat.yml
elasticsearch: http://192.168.137.121:9200...
  parse url... OK
  connection...
    parse host... OK
    dns lookup... OK
    addresses: 192.168.137.121
    dial up... OK
  TLS... WARN secure connection disabled
  talk to server... ERROR Get "http://192.168.137.121:9200": EOF
```

During this time, wireshark with `tcp.port eq 9200` captures:
[image]

------

machinebeat run log

```log
2022-04-08T22:12:28.557+0800	ERROR	[publisher_pipeline_output]	pipeline/output.go:154	Failed to connect to backoff(elasticsearch(http://192.168.137.121:9200)): Get "http://192.168.137.121:9200": EOF
2022-04-08T22:12:28.557+0800	INFO	[publisher_pipeline_output]	pipeline/output.go:145	Attempting to reconnect to backoff(elasticsearch(http://192.168.137.121:9200)) with 1 reconnect attempt(s)
2022-04-08T22:12:28.557+0800	DEBUG	[esclientleg]	eslegclient/connection.go:260	ES Ping(url=http://192.168.137.121:9200)
2022-04-08T22:12:28.557+0800	INFO	[publisher]	pipeline/retry.go:219	retryer: send unwait signal to consumer
2022-04-08T22:12:28.557+0800	INFO	[publisher]	pipeline/retry.go:223	  done
2022-04-08T22:12:28.557+0800	DEBUG	[esclientleg]	transport/logging.go:41	Completed dialing successfully	{"network": "tcp", "address": "127.0.0.1:1080"}
2022-04-08T22:12:28.558+0800	DEBUG	[esclientleg]	eslegclient/connection.go:264	Ping request failed with: Get "http://192.168.137.121:9200": EOF
2022-04-08T22:12:29.532+0800	INFO	nodevalue/nodevalue.go:217	[OPCUA] Publishing 15 new events
```

During running wireshark captures nothing with `tcp.port eq 9200`

---
I have been struggling with this for a long time, but have not found the reason, and I hope you can help me. Thank you!

BigeYoung, 2022-04-08:
Surprisingly, I used the exact same configuration on the same computer with Windows Subsystem Linux (copy and paste, absolutely identical!) Running machinebeat, it actually works!!!

felix-lessoer, 2022-04-08:
Thats looks really strange.
The connection to elasticsearch is used from the beat library and is used the same way in all the other beats.
All references I found for this issue appear to connect to the wrong endpoint.
In your case everything looks ok. 

Can I close the issue for you?

BigeYoung, 2022-04-08:
If this question is of no value to this repository, please close it.  
But I can also provide more information, including video, if you ask.

felix-lessoer, 2022-04-08:
Well its nothing Machinebeat can change.

I would recommend asking in https://discuss.elastic.co/

closed 2022-04-08 by felix-lessoer