elastic/Machinebeat #53: Issue sending OPC data to logstash from machinebeat

naupan, 2023-09-06:
I'm working on sending OPC data to a docker ELK stack and running into an error that seems to be related to the machinebeat input into logstash.  The attached screenshot is a portion of the logstash logs that seem to be pointing to the pipeline having an issue with input.  I'm not sure why the connection is being reset, or what the pipeline error could be.  I censored the remote address a bit.  That is the host running the OPC server, with machinebeat forwarding data to the docker logstash instance.

Any thoughts on what this could be?

[image]

felix-lessoer, 2023-09-06:
Which logstash version are you working on?

naupan, 2023-09-06:
It's 8.9.1.

naupan, 2023-09-07:
Additional info: OPC server is running on a Windows machine, so the latest version of machinebeat for windows is being used.  This is in a lab environment, so there are no SSL/certs configured between machinebeat, logstash, or the OPC server.  The ELK stack docker compose I'm using for testing is: https://github.com/deviantony/docker-elk.  The logstash pipeline config has no filtering, just the default inputs and output from the above linked docker compose.

felix-lessoer, 2023-09-07:
Thanks for your info. I will need some time to reproduce the issue within my environment.

felix-lessoer, 2023-09-08:
It looks like the beat framework version (7.8.1) thats used by machinebeat at the moment is not compatible with Logstash 8.9.1 . 
You can either install the 7.17.x version of logstash or wait until I can manage to update Machinbeat to the latest version.

What do you prefer?

naupan, 2023-09-08:
How much time would you anticipate an update taking?  I did some troubleshooting today and installed ELK 7.17.12 using 7.x branch of https://github.com/deviantony/docker-elk/tree/release-7.x

I wasn't able to get data from machinebeat to logstash still.  I've attached some of the logs after trying to change ecs_compatibility settings.  I tried disabled (default), v1, and v8.  No luck with any of those settings.

Were you able to push OPC data into logstash 7.17.12?  I attached my machinebeat and the opc module configs for reference.  

Attachments are sanitized slightly.  Not sure where I'm going wrong.  

[logstash_logs_ecs_compatibility_default.log](https://github.com/elastic/Machinebeat/files/12563463/logstash_logs_ecs_compatibility_default.log)
[logstash_logs_ecs_compatibility_v1.log](https://github.com/elastic/Machinebeat/files/12563464/logstash_logs_ecs_compatibility_v1.log)
[ymls.zip](https://github.com/elastic/Machinebeat/files/12563473/ymls.zip)

felix-lessoer, 2023-09-11:
I am able to reproduce the issue.
Now trying to find the cause for it.

felix-lessoer, 2023-09-11:
I think its not an issue of Machinebeat. What we see in this issue is related to the connection of Logstash -> Elasticsearch.
I was able to get the data send to Elasticsearch when the index is index named is defined like `index => "%{[@metadata][beat]}-%{[@metadata][version]}-%{+YYYY.MM.dd}"`
If the index configuration is not set then there is no data coming in. Even though it should be there in the logstash index.

The exception you see I only got when the beat is terminated. So this is most likely due to the fact that its usually a hard cut of the connection between the beat and logstash. However my logstash was still be able to receive a new connection from the next run of Machinebeat.

felix-lessoer, 2023-09-14:
My assumption is that there is an authentication issue with Elasticsearch.
For the beat direct to Elasticsearch connection you can see that its always rejected by Elasticsearch. 
For Logstash this error message seems also be related to the fact that it cannot connecct to Elasticsearch properly as well.

So please double check your connection / credentials / certificates (if https). Easy way to do that is to execute the following command on the host that is running Machinebeat.
`curl --user elastic:<not_real_pass> -X GET "10.200.x.x:9200/_cluster/health"`

Also if you cannot see it on your own, please send me your config files as well. 

Thanks for your patience

closed 2023-09-25 by naupan