elastic/Machinebeat #54: OPC UA Questions

SkyeControls, 2023-10-27:
Felix,

First off thank you for the OPC UA beat and the time and effort put forth. I am completely new to Elastic, Kibana, Beats etc.

I am a Automation/Robotics, and Windows developer, and have not really done other languages in depth. I installed the OPC UA beat and I have it working with one of our Skye Controls OPCUA IoT data tools. 

We current have a set of IoT tools for the automation world.

As you can see by the picture there are 2 sessions active. Is this normal?

[image]

Also here is my OPCUA.yml file contents.

# Module: opcua
# Docs: https://www.elastic.co/guide/en/beats/machinebeat//machinebeat-module-opcua.html

- module: opcua
  metricsets: ["nodevalue"]
  enabled: true
  period: 1s
  
  #The URL of your OPC UA Server
  #endpoint: "opc.tcp://milo.digitalpetri.com:62541/milo"
  #The URL of your OPC UA Server
  endpoint: "opc.tcp://xx.xx.xx.xx:51310/SkyeDataHub/DataAccess"

  #==========================  Security configuration ============================
  #username: ""
  #password: ""
  
  ##If you do not know which policies and modes are supported: start the beat in the debug mode and you will see which is supported by your server
  
  #Possible values: None, Basic128Rsa15, Basic256Sha256, Basic256, Aes128_Sha256_RsaOaep, Aes256_Sha256_RsaPss
  #policy: ""
  
  #Possible values: None, Sign, SignAndEncrypt
  #securityMode: ""

  ##Configuration for certificate based security
  #clientCert: ""
  #clientKey: ""

  ##This needs to be part of the certificates SAN URL and DNS-Name 
  #appName: "machinebeat"
  
  #==========================  Data collection configuration ============================
    
  ##Machinebeat now introduces Elastic Common Schema(ECS) compliant fields accross all data sources. For migration you can choose to include or exclude fields
  #ecsFields: true
  #legacyFields: false  
  
  ##The subscribe mode collects one event whenever there is a change of the value. This is the default.
  ## All collected events will be send after the configured period
  ## If you turn this off the beat will pull the current value after each period.
  #subscribe: true

  ##The browse mode is enabled at default
  ## This means, if the configured nodes have childs all subscribable nodes will be monitored
  #browse.enabled: true
  #browse.maxLevel: 3
  #browse.maxNodePerParent: 5
  
  ##How often we should retry to connect to the opcua server when something is failes after the first inital successful connection
  #retryOnError: 5

  ##How many collect threads should be opened. This is only relevant for non subscribe mode.  
  ## This limit can be reached when it takes longer to get a value than it is configured with period. 
  #maxThreads: 50
  
  #==========================  Node configuration ============================
  ##If this is not configured the browse will start at root. This is required if browse is set to false
  ## Configure the nodes directly to speed up start up of the beat
  #nodes:
  #-  id: "ns=2;s=Dynamic/RandomDouble"
  #   label: "Random Double"

My next question is, on our Skye OPCUA/DA tool we can browse and configure the ndoes as expose by the server. See picture.
[image]

How can I do the same within the OPCUA beat?

I am assuming I need to add something within the .yml file.

Also how many OPC tags can the beat handle, we have some applications that could have 2000 OPC Tags for just one location.

We also have the ability in our OPCUA server to return "BaseDataType" to a UA Client reading an "Any" datapoint.  BaseDataType is a valid UA datatype and it is capable of storing any value.  Can the OPCUA beat handle a "BaseDataType"

Thank you so much for all your help in advance.

Regards,
Skye Controls

felix-lessoer, 2023-10-27:
Hey Skye Controls,

thanks for reaching out.
The default configuration that you use should be able to browse all the available nodes and subscribe to every node that has an supported DataType. The beat will try to translate the opc-ua data types as good as possible.

You can configure the root node for browsing using the .yml file.
Based on your picture I assume that you wanna collect all the available data under `RDMStatus` and ignore the rest. In order to achieve the same within the beat you need to configure the RDMStatus node in the nodes part of the configuration. 

Hint: Every line that start with # is deactivated.
In order to configure root nodes you need to activate the `nodes:` line and provide at least one `id` . This node-id should be visible within your OPC UA browser. 

Caution: You obfuscated the endpoint of your opc-ua server in the config. But in the picture its still visible.

SkyeControls, 2023-11-01:
Felix,

 

Here is what I discovered. 

 

The OPCUA client does not add nodes that have “BaseDataType” associated with the tag.

 

Any of the tags referencing  “SkyeMQTT_Broker”  only get analyzed but not added. These tags are of “BaseDataType”.

 

2023-11-01T13:34:53.715-0400   INFO      nodevalue/client.go:426               Analyse node id ns=2;s=SkyeMQTT_Broker

2023-11-01T13:34:53.717-0400   INFO      nodevalue/client.go:426               Analyse node id ns=2;s=SkyeMQTT_Broker:lynchburg-elementary

2023-11-01T13:34:53.719-0400   INFO      nodevalue/client.go:426               Analyse node id ns=2;s=SkyeMQTT_Broker:lynchburg-elementary.mchs-ptcooler

 

These tags have the “Any” data type assigned from the OPCUA server. (See below).

 

 

 

These tags with conical datatypes get analyzed and added. (See below). Notice the direct datatypes without the “Any”

 

 

2023-11-01T13:34:53.697-0400   INFO      nodevalue/client.go:426               Analyse node id ns=2;s=RDD_Skye_PalletizerApp:M4063_Reads.PLC.PlcTypeData.ProcesserName

2023-11-01T13:34:53.700-0400   INFO      nodevalue/client.go:448               Add new node to list: ID: ns=2;s=RDD_Skye_PalletizerApp:M4063_Reads.PLC.PlcTypeData.ProcesserName| Type string| Name ProcesserName

 

 

Is there a way around this. Please let me know.

 

As always thank you for your help.

 

 

See below: 

2023-11-01T13:34:53.697-0400   INFO      nodevalue/client.go:426               Analyse node id ns=2;s=RDD_Skye_PalletizerApp:M4063_Reads.PLC.PlcTypeData.ProcesserName

2023-11-01T13:34:53.700-0400   INFO      nodevalue/client.go:448               Add new node to list: ID: ns=2;s=RDD_Skye_PalletizerApp:M4063_Reads.PLC.PlcTypeData.ProcesserName| Type string| Name ProcesserName

2023-11-01T13:34:53.701-0400   INFO      nodevalue/client.go:426               Analyse node id ns=2;s=RDD_Skye_PalletizerApp:M4063_Reads.PLC.PlcTypeData.ProducrCode

2023-11-01T13:34:53.701-0400   INFO      nodevalue/client.go:448               Add new node to list: ID: ns=2;s=RDD_Skye_PalletizerApp:M4063_Reads.PLC.PlcTypeData.ProducrCode| Type uint32| Name ProducrCode

2023-11-01T13:34:53.703-0400   INFO      nodevalue/client.go:426               Analyse node id ns=2;s=RDD_Skye_PalletizerApp:M4063_Reads.PLC.PlcTypeData.Revision

2023-11-01T13:34:53.710-0400   INFO      nodevalue/client.go:448               Add new node to list: ID: ns=2;s=RDD_Skye_PalletizerApp:M4063_Reads.PLC.PlcTypeData.Revision| Type string| Name Revision

2023-11-01T13:34:53.710-0400   INFO      nodevalue/client.go:426               Analyse node id ns=2;s=RDD_Skye_PalletizerApp:M4063_Reads.PLC.PlcTypeData.Status

2023-11-01T13:34:53.712-0400   INFO      nodevalue/client.go:448               Add new node to list: ID: ns=2;s=RDD_Skye_PalletizerApp:M4063_Reads.PLC.PlcTypeData.Status| Type uint32| Name Status

2023-11-01T13:34:53.713-0400   INFO      nodevalue/client.go:426               Analyse node id ns=2;s=RDD_Skye_PalletizerApp:M4063_Reads.PLC.PlcTypeData.VendorID

2023-11-01T13:34:53.714-0400   INFO      nodevalue/client.go:448               Add new node to list: ID: ns=2;s=RDD_Skye_PalletizerApp:M4063_Reads.PLC.PlcTypeData.VendorID| Type uint32| Name VendorID

2023-11-01T13:34:53.715-0400   INFO      nodevalue/client.go:426               Analyse node id ns=2;s=SkyeMQTT_Broker

2023-11-01T13:34:53.717-0400   INFO      nodevalue/client.go:426               Analyse node id ns=2;s=SkyeMQTT_Broker:lynchburg-elementary

2023-11-01T13:34:53.719-0400   INFO      nodevalue/client.go:426               Analyse node id ns=2;s=SkyeMQTT_Broker:lynchburg-elementary.mchs-ptcooler

2023-11-01T13:34:53.722-0400   INFO      nodevalue/client.go:426               Analyse node id ns=2;s=SkyeMQTT_Broker:lynchburg-elementary.mchs-ptcooler.batteryVoltageDragino

2023-11-01T13:34:53.724-0400   INFO      nodevalue/client.go:426               Analyse node id ns=2;s=SkyeMQTT_Broker:lynchburg-elementary.mchs-ptcooler.descriptor

2023-11-01T13:34:53.728-0400   INFO      nodevalue/client.go:426               Analyse node id ns=2;s=SkyeMQTT_Broker:lynchburg-elementary.mchs-ptcooler.deviceName

2023-11-01T13:34:53.730-0400   INFO      nodevalue/client.go:426               Analyse node id ns=2;s=SkyeMQTT_Broker:lynchburg-elementary.mchs-ptcooler.humidity

2023-11-01T13:34:53.732-0400   INFO      nodevalue/client.go:426               Analyse node id ns=2;s=SkyeMQTT_Broker:lynchburg-elementary.mchs-ptcooler.location

2023-11-01T13:34:53.735-0400   INFO      nodevalue/client.go:426               Analyse node id ns=2;s=SkyeMQTT_Broker:lynchburg-elementary.mchs-ptcooler.temp_probe

2023-11-01T13:34:53.738-0400   INFO      nodevalue/client.go:426               Analyse node id ns=2;s=SkyeMQTT_Broker:lynchburg-elementary.mchs-ptcooler.temperature

2023-11-01T13:34:53.741-0400   INFO      nodevalue/client.go:426               Analyse node id ns=2;s=SkyeMQTT_Broker:lynchburg-elementary.mchs-icbox

 

Regards,

Rick Lisowski

 

--------------------------------------------------------------------------------

From: Rick Lisowski

Skye Controls & Design Inc.

46 Tellidora Trail

Spencerport, NY 14559

Phone #: 585.293.7144

Fax #: 585.293.0502

Cell #: 585.739.3243

Website: www.SkyeControls.Com

Email: ***@***.***

All E-mail Virus scanned......

-------------------------------------------------------------------------------

CONFIDENTIALITY NOTICE

 

Nothing in this message is intended to constitute an electronic signature unless a specific statement to the contrary is included in this message.

 

The documents accompanying this transmission contain confidential information belonging to the sender which is legally privileged. This information is intended for the use of the individual or entity named above.  If you are not the intended recipient, you are hereby notified that any material in error please notify us IMMEDIATELY.

CONFIDENTIAL INFORMATION INTENDED FOR USE OF ADDRESSEE ONLY.

felix-lessoer, 2023-11-03:
Hey,

can you activate debug logging and share these logs with me as well?
The debug logs should contain some info about the dataType and why it was not choosen to collect from.

With that info I may be able to extend the dataType detection to also include the `any` datatypes.

Best regards
Felix

SkyeControls, 2023-11-03:
Felix,

 

I don’t know where to enable the debugging.

 

If you could tell me, I will gladly send the log files.

 

Thank you again for your help.

 

Regards,

Rick Lisowski

 

--------------------------------------------------------------------------------

From: Rick Lisowski

Skye Controls & Design Inc.

46 Tellidora Trail

Spencerport, NY 14559

Phone #: 585.293.7144

Fax #: 585.293.0502

Cell #: 585.739.3243

Website: www.SkyeControls.Com

Email: ***@***.***

All E-mail Virus scanned......

-------------------------------------------------------------------------------

CONFIDENTIALITY NOTICE

 

Nothing in this message is intended to constitute an electronic signature unless a specific statement to the contrary is included in this message.

 

The documents accompanying this transmission contain confidential information belonging to the sender which is legally privileged. This information is intended for the use of the individual or entity named above.  If you are not the intended recipient, you are hereby notified that any material in error please notify us IMMEDIATELY.

CONFIDENTIAL INFORMATION INTENDED FOR USE OF ADDRESSEE ONLY.

felix-lessoer, 2023-11-03:
Its this line that needs to get enabled.
https://github.com/elastic/Machinebeat/blob/c1799b4d7bb74d8cda2ef48105a4384044cbecd6/machinebeat.yml#L128

Your download should have something similar.

SkyeControls, 2023-11-03:
Felix,

 

Here is the error message I get.

 

 

 

 

Regards,

Rick Lisowski

 

--------------------------------------------------------------------------------

From: Rick Lisowski

Skye Controls & Design Inc.

46 Tellidora Trail

Spencerport, NY 14559

Phone #: 585.293.7144

Fax #: 585.293.0502

Cell #: 585.739.3243

Website: www.SkyeControls.Com

Email: ***@***.***

All E-mail Virus scanned......

-------------------------------------------------------------------------------

CONFIDENTIALITY NOTICE

 

Nothing in this message is intended to constitute an electronic signature unless a specific statement to the contrary is included in this message.

 

The documents accompanying this transmission contain confidential information belonging to the sender which is legally privileged. This information is intended for the use of the individual or entity named above.  If you are not the intended recipient, you are hereby notified that any material in error please notify us IMMEDIATELY.

CONFIDENTIAL INFORMATION INTENDED FOR USE OF ADDRESSEE ONLY.

 

https://github.com/elastic/Machinebeat/blob/c1799b4d7bb74d8cda2ef48105a4384044cbecd6/machinebeat.yml#L128

SkyeControls, 2023-11-03:
Sorry here is what I have…

 

 

Regards,

Rick Lisowski

 

--------------------------------------------------------------------------------

From: Rick Lisowski

Skye Controls & Design Inc.

46 Tellidora Trail

Spencerport, NY 14559

Phone #: 585.293.7144

Fax #: 585.293.0502

Cell #: 585.739.3243

Website: www.SkyeControls.Com

Email: ***@***.***

All E-mail Virus scanned......

-------------------------------------------------------------------------------

CONFIDENTIALITY NOTICE

 

Nothing in this message is intended to constitute an electronic signature unless a specific statement to the contrary is included in this message.

 

The documents accompanying this transmission contain confidential information belonging to the sender which is legally privileged. This information is intended for the use of the individual or entity named above.  If you are not the intended recipient, you are hereby notified that any material in error please notify us IMMEDIATELY.

CONFIDENTIAL INFORMATION INTENDED FOR USE OF ADDRESSEE ONLY.

 

 

 

 

 

 

 

 

 

 

 

https://github.com/elastic/Machinebeat/blob/c1799b4d7bb74d8cda2ef48105a4384044cbecd6/machinebeat.yml#L128

felix-lessoer, 2023-11-06:
Hey Rick,

unfortunately both replies are missing the essential piece. Can you try to upload it not via mail but via GitHub directly, pls?

Thanks Felix

SkyeControls, 2023-11-06:
Felix,

Here are the details from enabling debug mode.

[image]

Here is what I am enabling.

[image]

SkyeControls, 2023-11-06:
Sorry about closing it by mistake.

felix-lessoer, 2023-11-06:
You can keep logging.selectors as it is. So please add # again in front of that line. Make sure to not add any tab spaces or line breaks. 
It should work then.

SkyeControls, 2023-11-06:
Felix,

 

Here are the beats logs.

 

Regards,

Rick Lisowski

 

--------------------------------------------------------------------------------

From: Rick Lisowski

Skye Controls & Design Inc.

46 Tellidora Trail

Spencerport, NY 14559

Phone #: 585.293.7144

Fax #: 585.293.0502

Cell #: 585.739.3243

Website: www.SkyeControls.Com

Email: ***@***.***

All E-mail Virus scanned......

-------------------------------------------------------------------------------

CONFIDENTIALITY NOTICE

 

Nothing in this message is intended to constitute an electronic signature unless a specific statement to the contrary is included in this message.

 

The documents accompanying this transmission contain confidential information belonging to the sender which is legally privileged. This information is intended for the use of the individual or entity named above.  If you are not the intended recipient, you are hereby notified that any material in error please notify us IMMEDIATELY.

CONFIDENTIAL INFORMATION INTENDED FOR USE OF ADDRESSEE ONLY.

SkyeControls, 2023-11-06:
Felix,

Here are the logs.

[Beats Logs.zip](https://github.com/elastic/Machinebeat/files/13268375/Beats.Logs.zip)

felix-lessoer, 2023-11-09:
Thanks for sending the data. 
Unfortunatly this tells me, that there is no inbuild way to identify this data type automatically. 

The only work around that might work is to 
set subscribe and browse to false and configure the nodes with any datatype manually within the nodes section.

Note: In non subscribe mode the beat will actively read (pull) every time the configured period ends. So any value change in between will not be recorded.

Let me know if this works for you, pls.

SkyeControls, 2024-01-12:
> @SkyeControls <https://github.com/SkyeControls>  Can we close the issue or do you still need some help?

Felix,

 

You can close the issue, thank you for the help.

 

Regards,

Rick Lisowski

 

--------------------------------------------------------------------------------

From: Rick Lisowski

Skye Controls & Design Inc.

46 Tellidora Trail

Spencerport, NY 14559

Phone #: 585.293.7144

Fax #: 585.293.0502

Cell #: 585.739.3243

Website: www.SkyeControls.Com

Email: ***@***.***

All E-mail Virus scanned......

-------------------------------------------------------------------------------

CONFIDENTIALITY NOTICE

 

Nothing in this message is intended to constitute an electronic signature unless a specific statement to the contrary is included in this message.

 

The documents accompanying this transmission contain confidential information belonging to the sender which is legally privileged. This information is intended for the use of the individual or entity named above.  If you are not the intended recipient, you are hereby notified that any material in error please notify us IMMEDIATELY.

CONFIDENTIAL INFORMATION INTENDED FOR USE OF ADDRESSEE ONLY.

closed 2024-01-12 by felix-lessoer