elastic/Machinebeat #55: Code for PLC4x

meteoricfarm, 2024-01-12:
Dear Contributors,

First of all, Happy New Year!

I'd like to use PLC4x modules, but I cannot find any relevant code for explanation.
Does anyone guide me to use plc4x modules?

Sincerely,

felix-lessoer, 2024-01-12:
Hey @meteoricfarm 

thanks for considering Machinebeat. 
Can you elaborate a little more on what you would like to achieve?
Very general speaking you need to enable and change the plc4x config file to get it up and running.
plc4x offers various connectors to other protocols. There might be differences based on what you finally wanna use.

meteoricfarm, 2024-01-12:
Hi Felix,

I'd like to get values from the ModBus TCP server and send them up to Elasticsearch.

Sincerely,

felix-lessoer, 2024-01-15:
I tested using this config and a local modbus server

modules.d/plc4x.yml
```
# Module: plc4x

- module: plc4x
  metricsets: ["value"]
  enabled: true
  period: 5s

  #The URL of your PLC4X Endpoint
  endpoint: "modbus-tcp://localhost"

  #==========================  Node configuration ============================
  nodes:
  -  tag: "holding-register:1:REAL"
```

You need to change the endpoint and the nodes.

meteoricfarm, 2024-01-16:
Thanks for the reply.
I put a configuration file I used.

```
$ cat modules.d/plc4x.yml
# Module: plc4x

- module: plc4x
  metricsets: ["value"]
  enabled: true
  period: 5s

  #The URL of your PLC4X Endpoint
  endpoint: "modbus-tcp://localhost:5020"

  #==========================  Node configuration ============================
  nodes:
  -  tag: "holding-register:1:REAL"
```

The error like below.
```
$ ./machinebeat -c machinebeat.yml -e
2024-01-16T09:15:05.637+0900    INFO    instance/beat.go:697    Home path: [/home/vagrant/workspace/machinebeat] Config path: [/home/vagrant/workspace/machinebeat] Data path: [/home/vagrant/workspace/machinebeat/data] Logs path: [/home/vagrant/workspace/machinebeat/logs] Hostfs Path: [/]
2024-01-16T09:15:05.699+0900    INFO    instance/beat.go:705    Beat ID: 7e250205-10bb-4032-bb03-3ab1ebfe338f
2024-01-16T09:15:08.704+0900    WARN    [add_cloud_metadata]    add_cloud_metadata/provider_aws_ec2.go:79       read token request for getting IMDSv2 token returns empty: Put "http://169.254.169.254/latest/api/token": context deadline exceeded (Client.Timeout exceeded while awaiting headers). No token in the metadata request will be used.
2024-01-16T09:15:08.705+0900    INFO    [seccomp]       seccomp/seccomp.go:124  Syscall filter successfully installed
.
2024-01-16T09:15:08.708+0900    INFO    instance/beat.go:291    Setup Beat: machinebeat; Version: 7.17.9
2024-01-16T09:15:08.709+0900    INFO    [index-management]      idxmgmt/std.go:184      Set output.elasticsearch.index to 'machinebeat-7.17.9' as ILM is enabled.
2024-01-16T09:15:08.709+0900    INFO    [esclientleg]   eslegclient/connection.go:105   elasticsearch url: http://localhost:9200
2024-01-16T09:15:08.710+0900    INFO    [publisher]     pipeline/module.go:113  Beat name: vagrant-server
2024-01-16T09:15:08.710+0900    INFO    [monitoring]    log/log.go:142  Starting metrics logging every 30s
2024-01-16T09:15:08.710+0900    INFO    instance/beat.go:456    machinebeat start running.
2024-01-16T09:15:08.716+0900    INFO    [monitoring]    log/log.go:192  Total metrics   {"monitoring": {"metrics": {"beat":{"cgroup":{"cpu":{"id":"session-79.scope","stats":{"periods":0,"throttled":{"ns":0,"periods":0}}},"memory":{"id":"session-79.scope","mem":{"usage":{"bytes":137035776}}}},"cpu":{"system":{"ticks":20,"time":{"ms":27}},"total":{"ticks":70,"time":{"ms":77},"value":70},"user":{"ticks":50,"time":{"ms":50}}},"handles":{"limit":{"hard":524288,"soft":524288},"open":15},"info":{"ephemeral_id":"4a58466e-e5bd-4b95-a213-2e57b16408a4","uptime":{"ms":3100},"version":"7.17.9"},"memstats":{"gc_next":8563784,"memory_alloc":4409992,"memory_sys":29226248,"memory_total":10856992,"rss":45481984},"runtime":{"goroutines":32}},"libbeat":{"config":{"module":{"running":0,"starts":0,"stops":0},"reloads":0,"scans":0},"output":{"events":{"acked":0,"active":0,"batches":0,"dropped":0,"duplicates":0,"failed":0,"toomany":0,"total":0},"read":{"bytes":0,"errors":0},"type":"elasticsearch","write":{"bytes":0,"errors":0}},"pipeline":{"clients":0,"events":{"active":0,"dropped":0,"failed":0,"filtered":0,"published":0,"retry":0,"total":0},"queue":{"acked":0,"max_events":4096}}},"system":{"cpu":{"cores":16},"load":{"1":3.79,"15":0.35,"5":1.03,"norm":{"1":0.2369,"15":0.0219,"5":0.0644}}}}}}
2024-01-16T09:15:08.717+0900    INFO    [monitoring]    log/log.go:193  Uptime: 3.106967896s
2024-01-16T09:15:08.717+0900    INFO    [monitoring]    log/log.go:160  Stopping metrics logging.
2024-01-16T09:15:08.717+0900    INFO    instance/beat.go:461    machinebeat stopped.
2024-01-16T09:15:08.717+0900    ERROR   instance/beat.go:1026   Exiting: 1 error: metricset 'plc4x/value' not found
Exiting: 1 error: metricset 'plc4x/value' not found
```

I used Machinebeat-8.7.0 for linux release.
If you need more information, feel free to ask.
Sincerely,

felix-lessoer, 2024-01-28:
@meteoricfarm Sorry for the late reply.

There seems to be an issue with the cross compile into linux. 
On Windows this does work well. Do you have the chance to test in on windows?

meteoricfarm, 2024-01-28:
@felix-lessoer Thanks for the reply.

As you mentioned, machinebeat for windows worked.
But, other symtoms has been shown like below.
My modbus simulator's configuration decribed as below.

```
{
"server": {
  "listenerAddress": "0.0.0.0",
  "listenerPort": 5020,
  "tlsParams": {
    "description": "path to certificate and private key to enable tls",
    "privateKey": null,
    "certificate": null
    },
  "logging": {
    "format": "%(asctime)-15s %(threadName)-15s  %(levelname)-8s %(module)-15s:%(lineno)-8s %(message)s",
    "logLevel": "DEBUG"
    }
  },
"registers": {
  "description": "initial values for the register types",
  "zeroMode": false,
  "initializeUndefinedRegisters": true,
  "discreteInput": {
    "1": true,
    "2": false,
    "3": true,
    "4": false
  },
  "coils": {
    "1": false,
    "2": true,
    "3": true,
    "4": false
  },
  "holdingRegister": {
    "1": 52225,
    "2": "0xCC02",
    "3": 52227,
    "4": "0xCC04"
    },
  "inputRegister": {
    "1": "0xDD01",
    "2": 56578,
    "3": "0xDD03",
    "4": 56580
    }
  }
}
```

And my pcl4x.yml file described as below.
```
# Module: plc4x

- module: plc4x
  metricsets: ["value"]
  enabled: true
  period: 5s

  #The URL of your PLC4X Endpoint
  endpoint: "modbus-tcp://localhost:5020"

  #==========================  Node configuration ============================
  nodes:
  -  tag: "input-register:1:WORD"
  -  tag: "input-register:2:WORD"
  -  tag: "input-register:3:WORD"
[... 338 lines cut ...]
    "false, false, true, false, false, false, false, false, false, false, true, true, false, false, true, true"
  ],
  "plc4x.value.value.keyword": [
    "false, false, true, false, false, false, false, false, false, false, true, true, false, false, true, true"
  ],
  "sensor.id": [
    "holding-register:4:WORD"
  ],
  "sensor.id.keyword": [
    "holding-register:4:WORD"
  ],
  "service.type": [
    "plc4x"
  ],
  "service.type.keyword": [
    "plc4x"
  ],
  "_id": "oIZwUo0B9rZNVez7DNGu",
  "_index": "machinebeat-7.17.9-2024.01.28-000001",
  "_score": null
}
```

If you need more information, do not hesitate to ask.
Sincerely,

felix-lessoer, 2024-01-30:
Are you using an publicly available modbus simulator so that I can test against it?

meteoricfarm, 2024-01-31:
Yes, I am

I used https://github.com/cybcon/modbus-server with a 'test.json' configuration file in the repository.

gcmegmv, 2024-06-28:
Hi @felix-lessoer 

I am wondering why the module PLC4x seems to be missing on the repo but it is included on the binaries for the different OS. For example, it wouldn't be possible to use PLC4x with Docker.

closed 2025-09-15 by felix-lessoer