elastic/Machinebeat #57: OPCUA Int64 problem

MariuszTrojanowski, 2024-03-28:
Hello, 

we have issue with recognition of **Int64** datatype.

> 024-03-28T13:00:41.446+0100    INFO    nodevalue/client.go:426 Analyse node id ns=2;s=50221101.production.actParts
> 2024-03-28T13:00:41.474+0100    DEBUG   [Get DataType]  nodevalue/client.go:526 Was not able to detect DataType
> 

Is it possible to enable this kind of datatype?

ps. We found that this datatype is missing in client.go, func getDataType

`func getDataType(value *ua.DataValue) string {
	switch err := value.Status; err {
	case ua.StatusOK:
		switch v := value.Value.NodeID().IntID(); v {
		case id.DateTime:
			return "time.Time"
		case id.Boolean:
			return "bool"
		case id.SByte:
			return "int8"
		case id.Int16:
			return "int16"
		case id.Int32:
			return "int32"
		case id.Byte:
			return "byte"
		case id.UInt16:
			return "uint16"
		case id.UInt32:
			return "uint32"
		case id.UtcTime:
			return "time.Time"
		case id.String:
			return "string"
		case id.Float:
			return "float32"
		case id.Double:
			return "float64"
		}
`

felix-lessoer, 2024-03-28:
Hi, 

I can see that this datatype is implemented in the underlaying opcua library. 
So I think it should work if I add it to the list of getDataType with the same structure. 
However, I can not make it during easter holidays. But most likely afterwards. 

Best regards
Felix

MariuszTrojanowski, 2024-03-28:
Hi Felix, 
thanks a lot for information. It would be good if you could add this :)
Have Happy Easter!

Best, Mariusz

MariuszTrojanowski, 2024-04-09:
Hi Felix, 
I hope you are fine.
Did you have a chance to work with this topic?

Best, Mariusz

MariuszTrojanowski, 2024-04-11:
Hello All, 
the sollution is easy, just add to "module/opcua/nodevalue/client.go" in function "getDataType"
another case with int type and recompile program.

Best, Mariusz

felix-lessoer, 2024-04-11:
@MariuszTrojanowski sorry that you had to do it alone at the end . Its a busy time at the moment.

But glad to see that you got it working. Are you willing to contribute your change with a PR?