elastic/Machinebeat #59: OPC UA, how to set datatype for specific node

yongjuYoon, 2025-01-10:
Hello guys,

The datatype of the node I want to read is "BaseDataType", but I can not find the type, so the following error occurs.
- node info
[image]
- error log
```
2025-01-10T17:02:41.678+0900	INFO	instance/beat.go:291	Setup Beat: machinebeat; Version: 7.17.9
2025-01-10T17:02:41.678+0900	DEBUG	[beat]	instance/beat.go:317	Initializing output plugins
2025-01-10T17:02:41.678+0900	DEBUG	[publisher]	pipeline/consumer.go:148	start pipeline event consumer
2025-01-10T17:02:41.678+0900	INFO	[publisher]	pipeline/module.go:113	Beat name: samsung-l22
2025-01-10T17:02:41.678+0900	DEBUG	[registry.lightmodules]	mb/lightmodules.go:262	Light modules directory '%!d(string=/home/samsung/machinebeat/module)' doesn't exist
2025-01-10T17:02:41.678+0900	DEBUG	[modules]	beater/metricbeat.go:151	Available modules and metricsets: Register [ModuleFactory:[], MetricSetFactory:[mqtt/topic, opcua/nodevalue], LightModules:[LightModules:[]]]
2025-01-10T17:02:41.678+0900	INFO	[monitoring]	log/log.go:142	Starting metrics logging every 30s
2025-01-10T17:02:41.678+0900	INFO	instance/beat.go:456	machinebeat start running.
2025-01-10T17:02:41.679+0900	DEBUG	[cfgfile]	cfgfile/reload.go:132	Checking module configs from: /home/samsung/machinebeat/modules.d/*.yml
2025-01-10T17:02:41.679+0900	DEBUG	[cfgfile]	cfgfile/cfgfile.go:193	Load config from file: /home/samsung/machinebeat/modules.d/opcua.yml
2025-01-10T17:02:41.679+0900	DEBUG	[cfgfile]	cfgfile/reload.go:146	Number of module configs found: 1
2025-01-10T17:02:41.680+0900	WARN	[cfgwarn]	nodevalue/nodevalue.go:133	BETA: The OPCUA metricset is beta.
2025-01-10T17:02:41.680+0900	INFO	nodevalue/client.go:75	[OPCUA] Get all endpoints from opc.tcp://127.0.0.1:4840
2025-01-10T17:02:41.681+0900	INFO	nodevalue/client.go:85	[OPCUA] Your selected policy: http://opcfoundation.org/UA/SecurityPolicy#None and security mode: MessageSecurityModeNone
2025-01-10T17:02:41.681+0900	INFO	nodevalue/client.go:95	[OPCUA] Policy URI: http://opcfoundation.org/UA/SecurityPolicy#None with security mode MessageSecurityModeNone
2025-01-10T17:02:41.681+0900	INFO	nodevalue/client.go:103	[OPCUA] Set ApplicationName to machinebeat
2025-01-10T17:02:41.681+0900	INFO	nodevalue/client.go:105	[OPCUA] Set ApplicationDescription (SAN DNS and SAN URL) to machinebeat
2025-01-10T17:02:41.681+0900	INFO	nodevalue/client.go:118	[OPCUA] Set to anonymous login
2025-01-10T17:02:41.683+0900	INFO	nodevalue/client.go:129	[OPCUA] Connection established
2025-01-10T17:02:41.683+0900	INFO	nodevalue/nodevalue.go:176	Browsing is enabled. Data collection will start after discovery. Based on your server and browsing configuration this can take some time.
2025-01-10T17:02:41.684+0900	INFO	nodevalue/client.go:376	[OPCUA] Start browsing node: ns=1;s=TCPSpeed_R0
2025-01-10T17:02:41.684+0900	DEBUG	[Browse]	nodevalue/client.go:420	Start browsing at 
2025-01-10T17:02:41.684+0900	INFO	nodevalue/client.go:426	Analyse node id ns=1;s=TCPSpeed_R0
2025-01-10T17:02:41.684+0900	DEBUG	[Get DataType]	nodevalue/client.go:526	Was not able to detect DataType
2025-01-10T17:02:41.685+0900	DEBUG	[Browse]	nodevalue/client.go:486	Found 0 new nodes for browsing with ref id 0
2025-01-10T17:02:41.685+0900	DEBUG	[Browse]	nodevalue/client.go:407	Found 0 nodes to collect data from so far
2025-01-10T17:02:41.685+0900	INFO	nodevalue/client.go:409	Found 0 nodes in total to collect data from
2025-01-10T17:02:41.685+0900	DEBUG	[Browse]	nodevalue/nodevalue.go:181	Nodes to collect data from
2025-01-10T17:02:41.685+0900	INFO	nodevalue/nodevalue.go:186	Browsing finished
2025-01-10T17:02:41.686+0900	INFO	nodevalue/nodevalue.go:199	Found 0 nodes to collect data from.
```

Looking at [this answer](https://github.com/elastic/Machinebeat/issues/54#issuecomment-1803353885), it seems that there is a way to set the datatype directly
Is there any way to set datatype directly to ```opcua.yml```?

felix-lessoer, 2025-01-10:
It looks like you have already set up a dedicated node.
So I think the only change you need to make in opcua.yml is to set subscribe and browse to false.

Let me know if this works for you

yongjuYoon, 2025-01-10:
I'm already testing the case of reading a specific node.

It was successful to read the value for the node set to "Double".

Below is the opcua.yml file.

- opcua.yml
```
# Module: opcua
# Docs: https://www.elastic.co/guide/en/beats/machinebeat//machinebeat-module-opcua.html

- module: opcua
  metricsets: ["nodevalue"]
  enabled: true
  period: 5s

  #The URL of your OPC UA Server
  endpoint: "opc.tcp://127.0.0.1:4840"

  #==========================  Security configuration ============================
  #username: ""
  #password: ""

  ##If you do not know which policies and modes are supported: start the beat in the debug mode and you will see which is supported by your server

  #Possible values: None, Basic128Rsa15, Basic256Sha256, Basic256, Aes128_Sha256_RsaOaep, Aes256_Sha256_RsaPss
  #policy: ""

  #Possible values: None, Sign, SignAndEncrypt
  #securityMode: ""

  ##Configuration for certificate based security
  #clientCert: ""
  #clientKey: ""

  ##This needs to be part of the certificates SAN URL and DNS-Name 
  #appName: "machinebeat"

  #==========================  Data collection configuration ============================

  ##Machinebeat now introduces Elastic Common Schema(ECS) compliant fields accross all data sources. For migration you can choose to include or exclude fields
  #ecsFields: true
  #legacyFields: false  

  ##The subscribe mode collects one event whenever there is a change of the value. This is the default.
  ## All collected events will be send after the configured period
  ## If you turn this off the beat will pull the current value after each period.
  subscribe: true # true - period 기간 내, 모든 event를 모으겠다는 의미

  ##The browse mode is enabled at default
  ## This means, if the configured nodes have childs all subscribable nodes will be monitored
  browse:
  - enabled: false
    #maxLevel: 3
    #maxNodePerParent: 5

  ##How often we should retry to connect to the opcua server when something is failes after the first inital successful connection
  #retryOnError: 5

  ##How many collect threads should be opened. This is only relevant for non subscribe mode.  
  ## This limit can be reached when it takes longer to get a value than it is configured with period. 
  #maxThreads: 50

  #==========================  Node configuration ============================
  ##If this is not configured the browse will start at root. This is required if browse is set to false
  ## Configure the nodes directly to speed up start up of the beat
  nodes:
  -  id: "ns=1;s=TCPSpeed_R0"
     label: "TCPSpeed_R0"
#  - id: "ns=1;s=RobotSerialNumber_R0"
#    label: "col_rbtSn_r0"

```

i

yongjuYoon, 2025-01-10:
```
func getDataType(value *ua.DataValue) string {
	switch err := value.Status; err {
	case ua.StatusOK:
		switch v := value.Value.NodeID().IntID(); v {
		case id.DateTime:
			return "time.Time"
		case id.Boolean:
			return "bool"
		case id.SByte:
			return "int8"
		case id.Int16:
			return "int16"
		case id.Int32:
			return "int32"
		case id.Byte:
			return "byte"
		case id.UInt16:
			return "uint16"
		case id.UInt32:
			return "uint32"
		case id.UtcTime:
			return "time.Time"
		case id.String:
			return "string"
		case id.Float:
			return "float32"
		case id.Double:
			return "float64"
		}
	default:
		if value.Value != nil {
			logp.Debug("Get DataType", "Using default data type: %v", value.Value.NodeID().String())
			return value.Value.NodeID().String()
		}
		logp.Debug("Get DataType", "Default data type is not applicable")
	}
	logp.Debug("Get DataType", "Was not able to detect DataType")
	return ""
}
```

I think this function does not have a case corresponding to "BaseDataType". Is that right?

felix-lessoer, 2025-01-10:
Yes because BaseDataType can be anything but the Node is not telling us what it is in your case. 

Do you require browsing or are you fine adding the list of nodes you wanna monitor manually?

yongjuYoon, 2025-01-13:
I want to select and read a few specific nodes, but the node is set to 'BaseDataType', so I can not read the value when I use it according to the guide.

- something likes:
```
nodes:
  -  id: "ns=1;s=TCPSpeed_R0"
     label: "TCPSpeed_R0"
```

So, there is no way to get value for `BaseDataType`?
For example, , I understand that it can not be used in the following way, is that right?
```
nodes:
  -  id: "ns=1;s=TCPSpeed_R0"
     label: "TCPSpeed_R0"
    type: "Double" # <- example
```

yongjuYoon, 2025-01-13:
And i have one more question.
it seems to fail to get the value of array data from opcua, "Array type" is supported?

[image]